bitdump/BLHeli #381: problems with mamba f722 mini (packet loss and no current sensing)

m013, 2019-07-14:
tried  4.0.4 and 4.1 #1746
blheli versions Rev32.6.4 and Rev32.6.5 respectively

the first problem - huge packet loss (up to 40%) i cant even arm (rpmfilter warning).
with Rev32.6.5 two of the motors get no data

the second - although i have telemetry readings from the esc on osd (rpm and temperature), i have no current sensing. i know you have to apply some load on motors tjo get readings, but i did it via motor test tab and still nothing

[dump.txt](https://github.com/bitdump/BLHeli/files/3389717/dump.txt)

[image]
[image]

robschii, 2019-07-14:
i can not use  rpm filter    same problems
current sensor is only possilbe with the old and stable method
voltage reading is 18.1V by an full 4s lipo by using the esc sensor, with shunt method its okay

m013, 2019-07-14:
@robschii  , i was told that the problem was that mamba fc wouldnt spin all 4 motors with dma burst set to off. in my case only 2 were spinning. that means no rpm filter this time.

as for current sensor - i ve got similar readings (18.1 with 4s)

its better to avoid diatone's hardware

m013, 2019-07-17:
target specific part -

dma pin a09 1
dma pin a08 1
dma pin c09 0
dma pin c08 1

closed 2019-07-17 by m013