bitdump/BLHeli #383: TEKKO32 F3 METAL sometimes won't startup

ju-mpe-r, 2019-07-30:
From time to time some of TEKKO32 F3 METAL ESC's wont startup on arm(sometimes one, sometimes two ESCs).
Also LEDs on two of four ESC's not working.
What the problem can be?
ESC's are new.
Also tried 32.6.5 beta firmware.
---
When you can fix this or just give some clear info on this issue? Because I'm not only one with such kind of issues...

sskaug, 2019-07-31:
I brought this up with Airbot, and they recommend that the units are returned to the dealer for replacement.

ju-mpe-r, 2019-08-02:
But what is the problem? Bad hardware design, or what? Is it possible to solive it by myself?

ju-mpe-r, 2019-08-02:
Also people says that after replacement have the same issues...
And if esc's sometimes won't startup how it can be that the problem in hardware?
Also not working LEDs...
Can you please ask Airbot to make official statement somewhere and somehow? Where did you asked Airbot about theese problems?

sskaug, 2019-08-03:
If you contact me at blheli32 at gmail dot com, I can put you in contact with Airbot

ju-mpe-r, 2019-08-05:
> If you contact me at blheli32 at gmail dot com, I can put you in contact with Airbot

I send you email.
Thank you!

ju-mpe-r, 2019-08-14:
Airbot answer:

I hardly to trig the issue on your ESc, but on another one it is very easy to trig, 
When there is probelm , the 6V LDO output droped to 2.xx V, which is not enough for the MCU. 

My next step will be try to remove the parts from the ESC, one by one, to see which one casued the problem, 
it sounds like during initialization, the MCU triged wrong IO level that pulling the 3.3v down, then 6v protected after that. 

I am talking about software and hardware, becuase the wrong pull up or down, can be hardware or software, since it happens only on start, once you have it started, there is totoally no problem at all. 
So I am think it can be:

A during boot, before the MCU load the firmware, the IO is in uncertain status, it could casue that probelm 
B when MCU boot, when it setup the unused IO status, may somewhere went wrong, that pull the status to wrong HIgh or low

It should not releated to the LDO's output capability, becuase in all progress, the current is not high from the input side. 

If you wish, please do the following test:
Trig the issue, then measure the 3.3v and 6v voltage 

Regards
Li
---
So, this may be software problem. @sskaug , can you please check corresponding initialization for timeout) or maybe write check function?)

ju-mpe-r, 2019-08-18:
@sskaug, what you think about Airbot's thoughts?

ju-mpe-r, 2019-08-21:
@sskaug, I'm thinking about this - if sometime you can't write firmware anymore for some beyond you reasons... What will be with quadrocopter community? Did you think about that? Because even now, I see, that working with current issues is in freeze.

sskaug, 2019-08-22:
Well well.... At the moment this is on Airbot's plate for investigating hardware. If they want me to help look into firmware, I'll be happy to do that.

ju-mpe-r, 2019-08-30:
@sskaug please, read carefully what airbot wrote:

> I am talking about software and hardware, becuase the wrong pull up or down, can be hardware or software, since it happens only on start, once you have it started, there is totoally no problem at all.
So I am think it can be:

A during boot, before the MCU load the firmware, the IO is in uncertain status, it could casue that probelm
B when MCU boot, when it setup the unused IO status, may somewhere went wrong, that pull the status to wrong HIgh or low

It should not releated to the LDO's output capability, becuase in all progress, the current is not high from the input side.

>

ju-mpe-r, 2019-09-03:
@sskaug What you think about airbot quote?

ju-mpe-r, 2019-09-05:
@sskaug are you still working on blheli32 or project is going to die? You already got our money for your soft, but no support at all. It's ok for you?

sskaug, 2019-09-05:
We have checked code and followed up with Airbot on this and want to assist and have offered to assist. Please continue the dialog with them.

striker890, 2020-03-10:
Any follow up on this? I am faceing the same issue...

githubtd01, 2021-04-19:
I have similar issues with HolyBro Tekko32 F3 ESC (35A) ESCs and BLHeli_32.8. I bought them in January 2021.

Any solution for this?

githubtd01, 2021-04-19:
What I find hard to understand is why, if it happens, several ESCs are affected and not only one. So there should be another cause for that I think.

githubtd01, 2021-04-25:
@sskaug any news about this case, should I return the ESCs?

sskaug, 2021-04-25:
We are not aware of any firmware issue relating to this, so yes, I would recommend to contact the vendor about the issues.

githubtd01, 2021-05-01:
Thank you @sskaug. 

There seems to be an issue with the startup. It seems as if the ESCs which have the green LED activated are never affected.

Any hints what could lead to a freeze during the bootup?

githubtd01, 2021-05-01:
I have read in another forum that the ramp up power might be an issue. I have it set to 45. Should I go to 50 again? Could there be any connection here?  

Or should I add additional caps?
THX!

githubtd01, 2021-05-01:
Could it be that the ESCs change into boot mode as described in other threads?

Probably because of a signal timing issue, delayed FC startup or something else?

mathiasvr, 2021-05-01:
@githubtd01 What version of betaflight are you using? Try updating if not latest.
You can also try disabling dshot bitbang mode.
Does this happen with both normal and bidirectional dshot?

githubtd01, 2021-05-02:
Don't laugh, I'm using PWM and Arducopter... .
Even if I disable PWM output if the FC is not armed it does happen. The only pattern I have found so far is that it never happens for the ESCs which have green LEDs active/are connected to output 1 and 5.

mathiasvr, 2021-05-02:
Okay, I see. I wondered if it could be related to a dshot issue but seems to not be the case then.

githubtd01, 2021-05-03:
It is not the LED colors and not the ramp up power. I decided to return the ESCs. I guess HolyBro will say I have soldered them and there is no return. Lesson's learned: I stay with the ECSs from another manufacturer which do work without issues.