bitdump/BLHeli #384: Feature request: make an option to forcibly specify supported ESC protocol

voroshkov, 2019-08-07:
Recently different people with different hardware started experiencing intermittent self-arming issues after crashes or drones being exposed to water.
 
What if we could forcibly set the DShot protocol as the only one accepted by ESC in BLHeli settings? Would it make our drones less prone to self-arming in damaged condition?

WhiteWind4, 2019-08-07:
Many times seen this problem on the wet drones.

sskaug, 2019-08-07:
This is good input. Particularly the old pwm formats are susceptible to false detection.
At the moment we have a Rev32.6.6 test code out that we plan to release as a Rev32.7 full release in a few weeks, after proper testing, and I do not want to delay that or put that at risk by adding more functionality.
But after that rev is out, we can look into this.

voroshkov, 2019-08-07:
@sskaug, Great! Thank you!

mikeller, 2019-08-07:
Fully supporting this. We've had issues reported against Betaflight by users experiencing unrequested motor spinups when selecting Dshot1200, which on second thought could be caused by capacitive effects on the motor control line causing the Dshot1200 frame to be detected as a PWM pulse.