bitdump/BLHeli #385: Failed while flashing,and now esc not detected

emski0, 2019-08-12:
Hi
I was flashing the Tekko32 F3 escs( Furling32 Multi) with the latest beta. the first two escs flashed no problem, the 3rd esc failed while flashing, and now it is not detected anymore. Tried afterward to flash esc #4 and it failed also while flashing and stopped being detected afterwards.
How can reflash them or what should i do to recover them?

4712, 2019-08-12:
The best way should be: Power up FC (connect USB cable), select the right port and click "Connect", then power up the ESCs, (wait until all beeps have stoped) and click "Check".

emski0, 2019-08-12:
i tried it, but the 2 two escs that failed while flashing, are not detected anymore.
I tried also one of the esc with arduino via 4way-if, and still not detected.