bitdump/BLHeli #390: HAKRC_35A_Multi_32_64/66 ESC dead at midflight

singlag, 2019-08-28:
Log File : https://drive.google.com/open?id=1WDwDTGv4s8zX0oXvy-1rCkzUh0l7kYos
DVR : https://www.youtube.com/watch?v=trc7YJGf83g

Betaflight 4.1 with 32.66 (4.0 with 32.64)
HAKRC 35A ESC's, 1000uF low-ESR cap on the power rails
2300kv SunnySky 2305 motors, 4S battery, bidirectional DShot enabled.
PID and filter using default value

everything working normal, but ESC dead at midflight after a fast flip/roll
I have replaced a new ESC and motor, but it happen again today, and no issue at 32.60

singlag, 2019-08-28:
1 esc no connection, no startup tone, no led, no fire, no burn mark, it just dead silently

JBlocks, 2019-11-19:
I just found this error here. 
The issue also affecting me. Same behaviour with an Airbot F7 FC and Airbot Furling 32 4-in-1 F3 ESC V1.2 45A with BLHeli 32.64 (nightly) as well as 32.7 and Betaflight 4.0.x and 4.1.1. 

Your DVR (https://www.youtube.com/watch?v=trc7YJGf83g) shows the issue after a front flip. Same with my setup. Can make a lot of rolls (left and right), but a front flip ends with your results. The whole drone disarms and falls from sky.