bitdump/BLHeli #394: blheli32 Rev32.66 for typhoon_32 4 in 1 not found

greybush937, 2019-09-14:
I was starting the process of getting my rpm filters set up and hit a snag, I'm unable to find the hex file for my esc, im looking for Typhoon32_Multi_32_66.Hex if its available but only see one for 32_64. Will there be one released or should i role back to my previous version of BF?

sskaug, 2019-09-15:
You can find 32.6.6 here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/BLHeli_32%20Test%20code%20Rev32.6.4%20hex%20files/Rev32.6.6%20for%20inverted%20bidirectional%20dshot%20and%20GCR%20return%20data%20only

greybush937, 2019-09-15:
Thank you very much, this new interface and betaflight is really blowing my
mind, the devs have really stepped it up, and i cant wait to try the rpm
filter.
      T