bitdump/BLHeli #397: blheli32 Rev32.66 for HGLRC Forward 65A (STM32L431)

Mako2312, 2019-09-18:
Hey Guys,

i am looking for the Software to flash my esc for RPM Filter.
I cant find my ESC (blheli32 Rev32.66 for HGLRC Forward 65A (STM32L431)) in the List,
just another HGLRC Forward with another mcu.
Can you help me out theire.

sskaug, 2019-09-18:
Now posted

closed 2019-09-18 by sskaug

Mako2312, 2019-09-18:
@sskaug what you mean with: now posted?

sskaug, 2019-09-19:
I mean that this code: Forward_L4_ESC_Multi_32_66.Hex is now posted here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/BLHeli_32%20Test%20code%20Rev32.6.4%20hex%20files/Rev32.6.6%20for%20inverted%20bidirectional%20dshot%20and%20GCR%20return%20data%20only. Isn't this the code you need?

Mako2312, 2019-09-19:
Sorry i was blind!
Thx you very much for that 😊