bitdump/BLHeli #400: Eagle Tree MicroVector protocol 107 needed

diveflyfish, 2019-09-25:
Hello all,
I am attempting to use BLHeli pass through an ET MicroVector FC.  They state:
“James,

It appears that the latest BLHeli has dropped support for protocol version 106. We'll investigate upgrading to 107, but in the interim if possible, we suggest trying a BLHeli that uses version 106. “

How can I determine what variant release of BLHeli_32 uses a 106 protocol that would allow me to program my Tekko 32 Metal ESC’s for my hexacopter build?  Thank you for any input and suggestions.
Very Best Regards,
James

4712, 2019-09-26:
@diveflyfish: BLHeli_32 needs protocol version 107 from rev 32.0 on, means from the very start.
As long as ET is not supporting configuration of BLHeli_32, you might want to use an Arduino Nano flashed to a 4way-interface.

diveflyfish, 2019-09-26:
Thank you for the clarification.  Are you aware of any website where this variant may be archived?  I am not familiar with Arduino pass through but I will research it. If you point me in a good direction that would be super.  Just out of curiosity is there any reason why ET is not or will not support protocol 107?  Thank you again for your candor and communication.  
Very Best Regards

James E Hamman DDS

4712, 2019-09-26:
Maybe reading https://github.com/4712/BLHeliSuite/blob/master/Manuals/BLHeli%20programming%20adapters.pdf page 16-18 is a good first step. 
You can find the pinout at https://github.com/4712/BLHeliSuite/blob/master/Manuals/BLHeliSuite%204w-if%20interfaces%20pinout.pdf page 2.

diveflyfish, 2019-09-26:
Here is a question,
Is there any way for me to purchase or request a license for pass through for the MicroVector as a one off in case ET decides not to pursue 107 protocol? Or is it a specific firewall issue that Eagle tree needs to unlock before that could happen?  Is this even remotely possible or hypothetically so?
Is this a purchase issue for ET needing to purchase the rights of 107 etc?  
Thank you for your help again.
Very Best Regards,

James E Hamman DDS

4712, 2019-09-26:
The interface protocol revisions are only different in how the interface handles the ESC connection.
Version 107 added the support for BLHeli_32 ESC. The version number helps the software to know, which commands can be used with the interface and what types of ESC are supported.

As an example here is the first [commit ](https://github.com/4712/betaflight/commit/746040d6cdb3702107a706032c3aa5f3fae05ebd)from May 2017.

So nothing to purchase, there are only several lines of code and values to be change.

diveflyfish, 2019-09-26:
Dear BLHeli support,
Thank you for this information.  The coding then would need to be accomplished by the ET team that code their FC correct?
Or is there a back door means to do so aside from the Arduino intermediate method?
Very Best Regards,

James E Hamman DDS

4712, 2019-09-27:
Sure, it's a matter of code inside the FC firmware.