bitdump/BLHeli #402: Motor speed signal from uC pin?

AndrewRabbitt, 2019-10-04:
Does anyone know whether any of the pins on the SiLabs uC cycle with motor phase or rpm in such a way that a speed-related signal can be measured?

On the SimonK firmware PD3 is being cycled every phase rotation so the motor speed can be measured by an external device but I want to do the same think with a BLHeli-programed SiLabs ESC.  Any hints would be appreciated!

zobotek, 2019-10-04:
What I have done in the past is to find an unused pin, modify the code to
toggle it and submit a pull request.

AndrewRabbitt, 2019-10-07:
OK.  Thanks.  It looks like I'll have to get sophisticated and learn some assembler.

sskaug, 2019-10-07:
You can look in the Silabs code. Search for macros named Set_RPM_Out and Clear_RPM_Out.

AndrewRabbitt, 2019-10-10:
Thanks Steffen!

AndrewRabbitt, 2019-12-20:
OK time ot pick this project back up again, but also dumb question time...!

I have a ZTW Spider Pro 20A Premium (I think - F390 A00FCV 1613 on the CPU) which is my target device and uses a QFN24 package leaving what I think are at least two spare pins P2.0 and P2.3  Ideally it'd be really nice to recycle P2.4, if possible as it has a nice pad I can solder my signal wire to.

So, I think what I need to do is:  (correct me if I'm wrong)

-  add/modify a port definition in the ZTW....inc file in the Port 2 definitions section to give my signal port a label.
-  add a switch high command with my new port label into one of the Set_RPM macros in the main code and
-  add a switch low command into one of the Clear_RPM macros, preferably the next one along.
-  learn how to compile the code
-  attempt to flash this binary into my ESC and test.

Does this make any sense to people that know?

AndrewRabbitt, 2020-01-12:
So, I think I've had some success here now.  I've added the following to the ZTW_Spider_Pro_30A_HV.inc file, since this is what BLHeliSuite told me was already on the board:

> ...
> MACRO Set_RPM_Out
> 	setb	P2.DebugPin
> ENDM
> MACRO Clear_RPM_Out
> 	clr	P2.DebugPin
> ENDM
> ...
> 

where P2.DebugPin is hopefully the handy port going to the JTAG pads on the board.

I then used this handy how-to Youtube video on compiling the assembler using the Raisonance Toolchain.  https://www.youtube.com/watch?v=7Mwc762p4z8  It seemed to work OK - BLHeliSuite flashed it and the motor and ESC still runs!  Partial success at least!

Now I just have to solder a wire on there and measure to see if anything is coming out!  Fingers crossed!

AndrewRabbitt, 2020-01-12:
Well, a friendly loan of an oscilloscope shows that everything is working tickety-boo!  I'll shut this issue down now and leave it in the archives for posterity.

closed 2020-01-12 by AndrewRabbitt