bitdump/BLHeli #403: MacOS 10.15 Catalina Support

SebiWolze, 2019-10-08:
Hello DEVs!
The actual Version of BL_Heli32 for MacOS isn´t supported by the newest MacOS Version 10.15 (Catalina) 
32Bit Applications aren´t supported anymore.

[image]

4712, 2019-10-08:
@SebiWolze Please download [BLHeliSuite32xMacOS64_Beta_ 0.9.1.1.dmg.zip](https://github.com/bitdump/BLHeli/blob/master/BLHeli_32%20ARM/README.md)

abqsdfoster, 2019-10-10:
I have installed the "..OS64.." version twice and my Mac still claims it is 32 bit.

4712, 2019-10-10:
@abqsdfoster Maybe because 32bit and 64bit dmg files had the same name, and the mac archive manager unpacks the zip automatically after download, it was too easy to mix them up. 
Now the dmg names are different (BLHeliSuite32xm_MacOS64... and BLHeliSuite32xm_MacOS32...).
Please try again.

abqsdfoster, 2019-10-10:
Yes the dmg file names are different, however the name of the installed application still ends in 32. Renaming to 64 does not solve the problem either. Have you been able to replicate the problem on your Mac?

4712, 2019-10-10:
No, so far I could not replicate here...

satscottfoster, 2019-10-10:
[image]

Before upgrading to Catalina you are supposed to check for legacy software (32 bit). This software is not supposed to work after the upgrade.

tracernz, 2019-10-11:
That is why you need to download the 64-bit version, linked by @4712.

4712, 2019-10-11:
@satscottfoster
The MacOS system report collects information about legacy software and keeps it even if the folder or app was removed long ago. 
If you look at the "Last used" column, the date does not match the last date you used the latest BLHeliSuite32xm version last time.
Unfortunately, there seems to be no (easy) way to clean up such outdated information.

Looking for BLHeliSuite32xm under Software/Applications/ will list it as 64-bit = Yes.
So, more a cosmetic problem of the system report....

mikeller, 2019-10-12:
@4712: As a workaround, how about publishing the 64 bit version of the application under a different name, like 'BLHeliSuite32xm_64bit'?

4712, 2019-10-12:
@mikeller There is no practical problem, so nothing to work around here...

tracernz, 2019-10-12:
In any case, is there any point building a 32-bit macOS version? They only briefly shipped 32-bit x86 CPUs for a few months in 2006, and the last version of OS X to support them was Snow Leopard in 2009.

mikeller, 2019-10-12:
@4712:

> Unfortunately, there seems to be no (easy) way to clean up such outdated information.

This sounded to me like a workaround was needed for users who had installed the 32 bit version of the app and are now getting an error when trying to run the 64 bit version because of stale information about the app being cached.
But maybe I misread.

4712, 2019-10-13:
@tracernz 
> In any case, is there any point building a 32-bit macOS version? They only briefly shipped 32-bit x86 CPUs for a few months in 2006, and the last version of OS X to support them was Snow Leopard in 2009.

Very good point, I guess we can remove the 32bit macOS version...

4712, 2019-10-13:
@mikeller There are suggestions, before upgrading to Catalina, to check the system for 32-bit apps, which would stop working afterwards. Under Mojave one way to do so, is the read out the system report "Legacy Software", which lists also outdated info (under High Sierra this category is not listed at all). I don't think there is any real show stopper while doing the upgrade.

mikeller, 2019-10-13:
@4712: In this case, if a 64 bit app is shown as a 32 bit app in the report, this sounds more like a bug in MacOS.

4712, 2019-10-13:
@mikeller  [this ](https://www.mac-forums.com/forums/macos-operating-system/349511-legacy-software-system-information.html) and [this](https://eclecticlight.co/2018/10/13/dont-believe-system-informations-legacy-applications/#comment-37030) might bring some light...

As mentioned before:
> Looking for BLHeliSuite32xm under Software/Applications/ will list it as 64-bit = Yes.
> So, more a cosmetic problem of the system report....