bitdump/BLHeli #407: Blheli_32.7.0.2 Flash failed multiple airboth wraith32 v3 ESC's

adalta87, 2019-10-24:
I was connecting to the esc via uno atmega328. All was fine when connecting to the 4 esc and I was trying to update them to 32.7 using 4-way-if bootloader function they were all four connected to pins 8,9,10,11 on the uno. Flashing failed on ESC #1 and then it ended up in this constant flash failed sequence where I am unable to fix it tried just connecting a single esc to it and it also failed and again is stuck in flash failed sequence like the other, then tried one more for good measure and it to had the same results. When I make interface i am choosing the uno atmega328 and the arduino 4-way interface  and the multiv20003.hex and the baud rate is 115200. Am I doing something wrong, I have read about this magic c2 thing to fix bricked esc's but have not found much info about this. Have I screwed up my 3 esc's for good?

dlwalter, 2019-10-24:
We are also unable to flash 2-4S ESCs using 32.7.0.2.  We had no problems using 32.6

adalta87, 2019-10-24:
Where can you find the 32.6 blheli_32 gui, I have been only able to find the latest?

dlwalter, 2019-10-24:
We've had it installed on the computer.

dlwalter, 2019-10-24:
I realized its 32.62, not 32.6

4712, 2019-10-26:
Thanks for heads up about this issue and the log files!

It was timing issue in the Arduino 4way-interface, which by chance only occurred with the new version >32.7.0.1 and the UNO.  
Please use the just published version 32.7.0.3, which includes a fix. 

Sorry for the inconvenience and thanks again!

closed 2026-11-14 by 4O6FPV