bitdump/BLHeli #409: BLHeli_32 Suite forces upgrade

dlwalter, 2019-10-24:
Trying to use BLHeli_32 Suite 32.6 we cannot flash firmware and the software forces us to upgrade.  Is there a box we can check to prevent checking for updates?  Clicking cancel to the upgrade puts you in a dialog box loop where it keeps asking you to upgrade.

4712, 2019-10-26:
@dlwalter After important bug fixes, a forced upgrade before the next flash is sometimes essential.

dlwalter, 2019-10-28:
Please be aware, people may be using your software in a production environment and when you force an upgrade to a broken version (4way was broken on 32.7.0.2) you are throwing a pretty big wrench downstream.  

With most software you can simply revert to the last usable, stable version, but since it forces an upgrade this is not possible.  32.62 was obviously very stable and I doubt the forced upgrade was essential.