bitdump/BLHeli #410: Strange ESC initialisation with H7 Extreme FC

haslinghuis, 2019-10-28:
I have a Spracing H7 Extreme with 4 Turnigy Multistar 51A Opto ESCs (Siskin_ST_32_Plus). With latest BLHeli 32.7 and latest BF4.1 ESC initialisation is ok. But when I turn on bidirectional Dshot using the slider in the latest BF Configurator the ESC initalisation gets weird (out of sync, one after the other and the last two at the same time). Also users report roll of death issues. 
Weird thing is I have solved the weird ESC initialisation with an analog calibration (https://github.com/betaflight/betaflight/issues/9098#issuecomment-545585967)
But after reflashing the ESC I experienced the same weird out of sync initialisation and could not reproduce my earlier solution to fix it.

Betaflight support (using slack) suggested reporting this issue.

haslinghuis, 2019-10-29:
See https://github.com/betaflight/betaflight/issues/9103

haslinghuis, 2019-10-29:
With the AIkon AK32Pin Blheli32 / Brainfpv Radix Li on BF4.1 RC6 there are no issues.

sskaug, 2019-11-04:
I have tried to reproduce this, but so far have not been able to.
I've used an Omnibus F4 SD as well as a Matek F7.
FC output signal looks good on an oscilloscope.

haslinghuis, 2019-11-05:
@sskaug https://github.com/jflight-public/betaflight/issues/1#issuecomment-550054344

About your reply in attached issue, I think the out of sync init is not related to Blheli32 firmware but with some boot delay in the H7 specific code causing the out of sync esc init,

hydra, 2019-11-07:
@haslinghuis that's my suspicion too.  I think the init sequence in BF needs looking at, years ago I did a whole bunch of work in CF to make sure that the motor outputs were valid and stay valid as-soon as possible, for oneshot/pwm escs.  It's possible similar work needs to be done again now.

@sskaug Would you like an H7Extreme for testing?  If so drop me an email and let me know where to send it.  You could maybe check the timings from power on/reset and see if you find anything that you think would cause problems for the BlHeli firmware.  At least then we'd have something to work towards fixing in either BlHeli, BF or both.

hydra, 2019-11-07:
I captured some long traces using my scope of cold boot and reboot behavior using latest BF 4.1 release with all settings at default except that DSHOT300 was selected and bidirectional DSHOT was enabled.

Cold boot, ~3.4 seconds before DSHOT signals start, once they start they do not stop.

[image]

Reboot, using 'save' in configurator on configuration page to trigger reboot.  ~2.3 seconds.

Note that just before the reboot the dshot signals STOP, then RESTART (just before the first vertical white line).

[image]

Here's the same capture, but zoomed in as far as I could:

[image]

Here's a spike that occurs on every reboot or cold boot, it's very short, 34us.

[image]

FC's motor outputs are connected to my FC tester during this probing.  This probing was done using motor output 4, directly on the CPU pin which is on the corner of the cpu and easy to attach probe to.

@joelucid The spike likely needs some investigation, perhaps the GPIO level is not set correctly before the GPIO is enabled?   

@sskaug do you think that short spike is enough to make BLHeli spin a motor?

hydra, 2019-11-07:
I also switched the configuration from DSHOT to ONESHOT42 using the configurator and captured the result of Motor 4.  Notice there is no spike when using ONESHOT42.

[image]

So likely the DSHOT initialisation code needs updating to remove that spike.

sskaug, 2019-11-08:
@hydra Great that you look into this. The short pulse could cause motors to spin during a reboot (if an appropriate signal was detected before the reboot). But it will not be enough to pass a signal detection.
For BLHeli, it is important that the signal line is held low until signalling starts, as the code will enter bootloader if input is high for a prolonged time (~15ms for BLHeli_32).

hydra, 2019-11-09:
@sskaug thanks for the response, to clarify can you confirm that:
a) the very short spike above is not long enough to trigger entering bootloader as it's much short than 15ms.
b) on cold boot the short spike will not spin the motors.
c) on warm reboot the short spike could spin the motors.  If so what motor protocols are affected? not DSHOT right as it's clearly not a valid frame.

also, please remind me if BlHeli ESCs need to be power-cycled before they will re-detect a different protocol once they have detected a valid signal.  i.e. you can't switch from say PWM to OneShot or OneShot to DSHOT without powering them off?

sskaug, 2019-11-10:
Sure, the very short pulse that you see will not be capable of invoking bootloader. And before signal is detected (beep number four), such a short pulse will not spin motors. On a warm reboot, such a pulse can at least theoretically spin motors, but should not do that for Dshot signal.
BLHeli ESCs will consider signal to be lost after a given time of no valid frames (quite long, in the order of a few hundred ms in order not to fail in the air). After signal is lost, the code will try to detect signal again (and still accept any type of signal).

hydra, 2019-11-15:
@sskaug Hi,

@haslinghuis kindly sent me his quad for further investigation and I can confirm his motor initialisation is indeed very odd.

The initial signal level after power up on the ESC line is HIGH, it then drops LOW, then the FC boots more, then a DSHOT signal appears, then a bit later the ESC's make the initialisation tone again.

Here's some scope traces:

1. power-on HIGH.

[image]

2. fc boot

[image]

3. fc boot zoomed

[image]

4. before second init tone.

[image]

5. after second init tone.

[image]

It would suggest the ESCs are entering bootloader mode and then timing out right?

hydra, 2019-11-15:
Here's a photo of Mark's quad, diff and dump.

[image]

[dump-all.txt](https://github.com/bitdump/BLHeli/files/3852192/dump-all.txt)
[diff-all.txt](https://github.com/bitdump/BLHeli/files/3852193/diff-all.txt)

hydra, 2019-11-15:
The SPRacingH7EXTREME has two buttons on it, one for BOOT (ROM DFU) and one for BOOTLOADER/BIND functions.

I note that if either button is held while power is applied that the signal level on the ESC motor input is LOW.

This means that neither the ST's ROM DFU bootloader not the SP Racing H7 botloader initialise the GPIO pins, so the problem is for sure in the BF firmware.

hydra, 2019-11-15:
The previous comment was behavior with battery DISCONNECTED.

However.

If you hold the BOOT button on boot, so stop the FC firmware booting, then connect the battery and measure the ESC signal pin it is HIGH.

Further more if you enter SP Racing bootloader mode (pressing and hold the BOOT button, wait for RED led to flash fast, release button). And measure the signal it is also HIGH when the battery is connected.  If you then press and release the boot button the bootloader will load the firmware and boot it, and the signal level goes low.

This seems to indicate that perhaps the ESC has a pull-up resistor on the motor signal input pin and/or the BL heli software is pulling the line high.

By looking at the scope traces above we can see the signal is high for 1.460 seconds.

Thoughts on how to proceed given this information requested.

hydra, 2019-11-15:
Mark also give me details of the ESCs and motors used:

ESC: 
* Turnigy MultiStar BLheli_32 ARM 51A Race Spec ESC 2~6S (OPTO)
* https://hobbyking.com/en_us/turnigy-multistar-32bit-51a-race-spec-esc-2-6s-opto.html?___store=en_us 

Motor:
* Turnigy Multistar 4220-880Kv 16Pole Multi-Rotor Outrunner
* https://hobbyking.com/en_us/turnigy-multistar-4220-880kv-16pole-multi-rotor-outrunner.html

hydra, 2019-11-15:
I guess some other information might be useful here.

The SPRacingH7 series FC's don't boot the firmware from internal CPU flash, they boot from an external storage system (EXST bootloader).  While the bootloader is detecting flash chips and looking for and loading files from SD card there can be a short delay.  The bootloader does not interfere with the motor GPIO lines, it only uses GPIOs for sdcard/quadspi/led/buttons.

As such if the ESC firmware samples the motor signal line, as boot/reboot to see if it's high, which it would be if there's a pull-up as the FC hasn't initialised the GPIO pin, then the ESC would enter bootloader mode, and then timeout.

On further investigation of mark's quad the SD card detect pins are not working so the bootloader tries to read from the SD card which makes the delay longer than it should be.  If an SD card is inserted the bootloader can read from it, find there is no firmware and then boots from external flash and the boot delay is reduced to 0.420 seconds and the ESCs don't do have the observed double initialisation behavior.

See scope trace below.

[image]

hydra, 2019-11-15:
@sskaug Can the ESC firmware be adjusted so that if the line is HIGH without communication, and then stays LOW for some period of time, that the ESC exits boot loader mode?

It feels like the issue can be bypassed with improved ESC bootloader timeout code?

joelucid, 2019-11-15:
@hydra yeah the way we’ve designed it is that since the esc pulls the line high the FC needs to drive it low as soon as it starts. We have a preinit function which does this both in bitbang and timer dshot. So there must be some other issue specific to h7 which prevents this from happening early enough. Probably early initialization taking longer than on the other boards. Or maybe the preinit needs to be called even earlier on h7.

hydra, 2019-11-15:
@joelucid The pre-init can't be called much earlier that it currently is called in the FC firmware.  IIRC it's already pretty early.   The delays in this case come from reading from the SD card.

The SP Racing BL firmware tries 10 times to read the SD card, with a 100ms pause between each attempt but only when SD card switch is closed by an SD card.  There are a lot of sub-par SD cards out there with long startup times before they respond, the BL needs to be compatible as possible.

The issue isn't H7 specific, for instance if I put my bootloader code on the SP Racing F3 and loaded firmware from the external flash, or on the SP Racing F3/F4 EVO or F7 DUAL with it's SD card socket. on that the same issue would occur.  The H7ZERO and H7EXTREME just happen to be the first widely shipped FC's that allow booting from SD card.

So IMHO, the issue isn't CPU related, it's ESC boot sequence related when no FC code is running or connected at all.

As I understand it, based on this thread, I think the ESC boot sequence should be something like this:
A) ESC power on or ESC reboot.
B) if signal is high wait for B1 or B2.
B1) a bootloader command, timeout after X seconds, jump to D.
B2) signal to go low and stay low for a period of time (say 200ms, but less than the time the motor signals start to appear). if signal is still low, and *never* went high again, abort bootloader wait sequence, jump to D.
C) if signal is already low jump to D.
D) detect motor signal protocol.
then usual behavior:
E) process motor signal and detect lack of motor signals for some period of time.
F) if no motor signal jump to D.

hydra, 2019-11-15:
I should also mention that the clock initialisation code in BF also has some code which waits for a while and reboots if it doesn't work, and that happens before the GPIO clocks are enabled which can also be a source of cold-boot long delays prior to motor signal going low.

See the implementation of `HAL_RCC_OscConfig` and note the default value of `HSE_TIMEOUT_VALUE` and `HSE_STARTUP_TIMEOUT` for each MCU.

sskaug, 2019-11-16:
@hydra Great failure analysis. Unfortunately we are very constrained in what we can do on the ESC side, since bootloader code is flashed at mfg time and can not be updated (for BLHeli_32 - for BLHeli_S it can, but it is not practical).
So we are limited on what we can do on the ESC side here.
But what about not accepting BL entry until xx seconds after powerup and lost signal?

hydra, 2019-11-16:
@sskaug yes, that seems like a reasonable solution.

Can you explain/clarify the proposed ESC boot process?

sskaug, 2019-11-16:
I think there are two options.
1. We do not go into the bootloader until a given time (say about 3 seconds) after powerup or loss of signal. This must be implemented very carefully, because we can not risk that the bootloader can not be accessed (in case of e.g. failure during flashing).
2. We require the signal line to be high for a very long time (say more than 3 seconds) before we go into bootloader mode. This will add a delay to entering bootloader, but should be very safe from an implementation point of view.

hydra, 2019-11-16:
how about a combination of 1 and 2. 

a 2 second MINUMUM delay seems reasonable based on what we currently know about startup behaviour (slow osc init + sd card search timeout).

it would be good to document the current and future behaviour somewhere too, and link to it from this issue.

bear in mind that boot loaders

mikeller, 2019-11-16:
@sskaug, @hydra: I suspect that we are going down the rabbit hole with this. I think we need to take a step back and consider the facts:
- the reported problem (ESCs not always initialising at the same time) seems to be cosmetic, and there is no report of it affecting flight performance or usability;
- there is a secondary problem reported (roll of death issues), but the assumption that this seems to be caused by inconsistencies in initialisation times seems to be far fetched;
- only < 0.15% of users use hardware that is affected by this. (Numbers are from Betaflight, I don't know what other firmware types support SPRacing H7 boards at the moment.)

So I don't know if it is the right approach to change anything that will add to the inconvenience of 99.85% of users by increasing wait / reboot times when using the BLHeli Suite just to address a cosmetic issue affecting 0.15% of users.

hydra, 2019-11-17:
I agree that the 'roll of death' issue is likely unrelated to this.

Updating bootloader/esc firmware now, prior to more boards being shipped, seems prudent otherwise we will all continue to field more support requests regarding it.

It is certainly the right approach to work together to resolve the issue.  As noted it's not H7 specific.  I'm also aware of other FCs, betaflight and otherwise, that will also be affected by this issue but I'm under NDA and can't give specifics.

I don't see how the proposed solution will cause problems for other users since the FC will still be booting and takes longer to boot than the proposed delay, and there's also the gyro calibration which also takes some time before arming is allowed so there's plenty of time for ESCs to finish initialising prior to arming.

proggod, 2019-11-17:
We have definitely seen initialization issues in the past with FlightOne which actually caused some catastrophic behavior, like full throttle.  I know that was fixed by waiting longer for multishot, however I highly support anything that will make initialization safer as long as it doesn’t delay booting. :)

hydra, 2019-11-17:
When it comes to ESCs safety and consistency is key.  We need to be able to say to people 'this is how it works', when it doesn't work as expected, as is the case with delays before FC code sets outputs low, then users get confused.

When users get confused:
* they raise support tickets, increasing burden on maintainers.  Additional support tickets also have the unwanted impact on the support systems and forums as a whole as they de-value them as they get cluttered.  We waste time answering them, redirecting them and so on, when our valuable time is better spend doing other things.
* they loose confidence in the models they fly and the software they use.  if they loose confidence they will be less likely to fly and more likely to complain.

Improving the startup behavior seems like a quick win now and for the future.

mikeller, 2019-11-17:
@hydra:

> Improving the startup behavior seems like a quick win now and for the future.

Not sure if I would consider a change that fixes something for 0.15% of users and makes it worse (slower) for 99.85% of users an 'improvement'. Statistically this is making it worse, and will most likely lead to a lot more support tickets from users complaining about this suddenly taking longer, just based on the fact that for every user that this provides a fix for there 600 users the proposed change makes the behaviour worse.

hydra, 2019-11-17:
@mikeller I think you mis-understood the problem and/or the solution.  As it is now the FC doesn't produce motor output signals until around 2 seconds after power on, and if it did that would still be fine.

The proposed solutions will have no impact as the motor output signal will already be low and thus normal ESC signal detection will catch the start of the motor signal output.

There is no additional delay to MOTOR detection, only ESC BOOTLOADER ENTRY.

hydra, 2019-11-17:
Now:

Bootloader case:
```
____________              __________
|          |______________|  
A          B     C        D
```

Normal FC case:
```
___                       __________
| |_______________________|  
A B              C        D
```

A) Power on *AND* ESC check for BOOTLOADER ENTRY.
B) Early FC init
C) ESC looks for motor signals (might happen earlier, but doesn't matter as FC isn't generating them yet).
D) Start of motor signal generation.

PROBLEM: ESC checks for bootloader entry on power up.

Proposed:

Bootloader case:
```
____________              __________
|          |______________|  
A          B  C  D        E
```

Normal FC case:
```
___                       __________
| |_______________________|  
A B           C  D        E
```

A) Power on
B) Early FC init
C) ESC check for BOOTLOADER ENTRY
D) ESC looks for motor signals (might happen earlier, but doesn't matter as FC isn't generating them yet).
E) Start of motor signal generation.

SOLUTION: ESC doesn't check for bootloader entry until C.

Notes:
* The time between power on and ESC starting to detect signals is the same now and for the proposed solution.
* The time between power on and FC generating signals is the same now and for the proposed solution.
* Bootloaders increase the time between A and B.

joelucid, 2019-11-17:
@hydra i think the potential issue is this probably most common scenario: user connects quad to usb and starts blheli32 configurator. At this point Bf (And blheli32) pull up the dshot line. Then user connects LiPo and clicks “read setup”. If he does so before the 2s timeout after connecting the LiPo the read would fail and he would have to retry.

Not sure if 2s is long enough to be problematic here. But this scenario is the one we should try to make work without 2 attempts. 

I think we should arrive a lot a solution that works for both existing and new hw. Arguing that next generation stuff is only used by few people is a sure path to become the next cobol.

hydra, 2019-11-17:
@joelucid yes, The scenario you mention isn't time-sensitive, throwing a quad on the ground and getting in the air is, and happens far more frequently than ESC firmware upgrades or ESC configuration right?  I had already thought of that scenario, but don't think it's an issue.

And heh, I'd be surprised if anyone can connect a battery connector, which normally needs two hands, one for each male + female connector, and then move their hands from them onto a mouse or keyboard and use on some UI element within 2 seconds, but the kidz today are quick! 😄 

I'm fine with a period of 1.75 instead of 2.0 seconds, if you think extra the 250ms would be useful in this scenario.  As measured above, the worst-case scenario is slow startup + sub-par SD card and that was about 1.5 seconds.

joelucid, 2019-11-18:
@hydra I don’t see the flying the quad as time critical as the scenario I described. Who arms 2s after connecting a LiPo? Of course configuring escs happens less often for normal people. I do it all the time :D. But I suspect even 2s is not something one would beat unless consciously trying. 

On the other hand if escs only start after playing the initial startup tones (which blhelis/jesc do) we probably want to subtract the startup tones from the 2s period.

mikeller, 2019-11-18:
@sskaug: Shouldn't it be possible to do the check for a boot loader request (signal line high) while at the same time detecting a stable signal? This should make it possible to implement this without adding additional delay for any users:

Normal FC case:

```
___              __________
| |______________|  
A B  C        D  E
```

Bootloader case:

```
____________              __________
|          |______________|  
A    C     B  D           E
```

A) power on
B) flight controller initialises motor outputs
C) ESC starts checking for input signal
D) if no motor signal has been detected, ESC checks for boot loader request
E) flight controller starts generating motor signal

sskaug, 2019-11-18:
I have now posted a test code here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/BLHeli_32%20Test%20code

hydra, 2019-11-22:
@sskaug I'll try and find some time to test that on the ESCs that @haslinghuis gave me this weekend.

hydra, 2019-11-23:
@sskaug  Ok! I flashed that firmware and it seems to fix the bootloader timeout issue on initial power-up.

All the ESC's play the startup tones, one is slightly delayed. (rising three beeps). Then there is a pause while the FC boots, then the remainder of the startup tones play (rising two beeps).

I note that if start BlHeliSuite32, connect, read setup and then disconnect, the same long delay occurs while the ESCs timeout of 'bootloader mode' as before.

After downgrading ESC 1 to 32.70 and leaving the rest on 32.71 the old behavior of very-delayed motor beeps occurred again only on ESC 1.

I also note that if I have power applied, start BLHeliSuite32, connect, read setup, all 4 escs are read, if I then power off the ESCs but keep the FC powered via usb, then power on the ESCs and click 'Check' it _always_ fails with 'Failed to read Flash!', at which point if you click 'OK' it shows 'Information' dialog with Motor 1 as 'Erased or unknown firmware' and ESC's 2-4 as normal.  If you then click 'OK' on the dialog it automatically retries and shows all 4 ESCs in the 'Information' dialog again.   This happens regardless of attempting to 'Check' immediately after power on or a little later.  I downgraded ESC 1's firmware to 32.70 (from the 32.71 test code above) and the behavior is the same, so I suspect that is unrelated.   Should I raise a new issue for this?

Can you share the ESC exact startup behavior with us? that is implemented in 32.71?

hydra, 2019-11-23:
Here's a video of the ESCs powering up from cold:

https://www.youtube.com/watch?v=tzO5WGuqzkk

There is audio, it's just very quiet on these motors.

4712, 2019-11-23:
> I also note that if I have power applied, start BLHeliSuite32, connect, read setup, all 4 escs are read, if I then power off the ESCs but keep the FC powered via usb, then power on the ESCs and [...]

[image]
[image]
[image]
[image]
And again
[image]

@hydra For sure a cosmetic issue of the logic.
I will try to improve it in the next revision.

hydra, 2019-11-23:
@4712 yes it was similar to that, except that for ESC 2-4 it detected the ESCs, just not for ESC 1.  If you need exact screenshots or details please create an issue somewhere and link to it from here, then I can add to it as required.

sskaug, 2019-11-23:
@Hydra Thanks for testing. In that 32.71 test code, I use a hardware timer from powerup to set a timewindow where BL is not entered. In addition to that (to make sure we do not for some reason have issues with partly flashed ESCs etc, I also added a pure wait time. If the code keeps waiting with input signal high, it will enter BL regardless of hardware timer, after about 3 secs.

4712, 2019-11-23:
> @4712 yes it was similar to that, except that for ESC 2-4 it detected the ESCs, just not for ESC 1. 
@hydra got it now, it occurs after power off/on cycle... thanks!

hydra, 2019-11-23:
@sskaug ok, let me know if you need me to test anything else for you.  Will you be adding this to the next release of BlHeli?  Do you need more testers?

@4712 great!

sskaug, 2019-11-24:
Thanks for the offer. We will add this to the next test code, so that wee will have some exposure before going to the next release. At the moment we are looking at adding support for SBUS/S.PORT, so we may have a test code for that in not too long.

haslinghuis, 2019-11-29:
@4712 same experience - you do not have to open a new issue. Please keep using this issue. It seems all related. Just timing {out};

hydra, 2019-12-13:
Any update on this?

sskaug, 2019-12-14:
The test code is published. Not much info neither on people having the issue nor whether it is resolved

spracingofficial, 2020-12-09:
@hydra What's the current status of this?

haslinghuis, 2021-06-17:
As I have sent my quad for testing (USB connector was broken off - as this was an early production problem) - a 10" frame with FC (H7 Extreme), Turnigy Multistar BL32 51A Race Spec ESC's and 4220-880KV 16 pole outrunner motors to @hydra I can't confirm myself. Happy to know what's the status.

haslinghuis, 2021-06-17:
Would like to test PX4 on this board. I have some expensive sensors (RP-Lidar A2M8, Garmin Lidar-lite V3HP, Px4flow sensor)

haslinghuis, 2022-04-07:
There is no strange initialization anymore, using BLHeli32 32,8,3 (didn't try 32.9.x).

Closing as resolved.

Rock solid now using BF 4.3 RC4 (see confirmation on https://github.com/betaflight/betaflight/issues/9886).

Not sure if this was the resolution: https://github.com/betaflight/betaflight/pull/11494 as a complementary fix is still waiting: https://github.com/betaflight/betaflight/pull/11469

closed 2022-04-07 by haslinghuis

Denethor85, 2024-03-25:
In my build, with the Rush Blade Sport 50A ESC and firmware 32.10 I suddenly have strange problems with the ESC not starting like in this thread. I had never had this with previous firmwares. If I go back to 32.9 or 32.8 the problem goes away. The problem is a continuous restart of the ESCs with desynchronized sounds

hydra, 2024-03-25:
@Denethor85 It's probably best to open a new issue and detail your exact setup (esc, firmware, esc config, fc, fc firmware, fc config, etc) and refer to this issue in your new issue.  You can also try the various RC/FPV/ESC/FC discord servers for trouble-shooting.