bitdump/BLHeli #411: Flash failed - ESC not detected in USB/COM

MrLatebros, 2019-10-29:
I have just got a Razor32 V2 35A ESC. It came with firmware ver. 36.6.
Connected it using the 4-way arduino interface. Writing any values failed. I tried to flash the update FW (36.7), it got about 40% there, then "flash failed".

I tried power-cycling the ESC, etc, now it's stuck in the flash corruption-please flash-flash failed loop.
I saw this issue: https://github.com/bitdump/BLHeli/issues/407
in which they have the same problem as me. I tried the advised fix of using the USB/COM interface with pin D11, but this does not detect the ESC when connected.

This is all with BLHeliSuite32 version: 32.7.0.3 , which claims to fix the isue. This is apparently not the case.

Can anyone advise? Maybe I am using the wrong pin on the arduino nano?
Either way, my ESC firmware remains corrupted.

4712, 2019-10-29:
@MrLatebros What Arduino do you exactly use?

MrLatebros, 2019-10-30:
@4712 Arduino Nano flashed as USB/COM interface.

4712, 2019-10-30:
@MrLatebros which chip is used on the Nano for the onboard USB/UART? (CH340?)

MrLatebros, 2019-10-30:
@4712 yeah, CH340

4712, 2019-10-30:
@MrLatebros I still cannot reproduce the failure here, neither with Nano CH340. 
Strange... The failure always occurs when reading the echo (because it is a 1wire interface) back.
How does the log look like, when using 4way interface?

Also: Could you check with 32.7.0.0?
[BLHeliSuite32_32700.zip](https://github.com/bitdump/BLHeli/files/3789075/BLHeliSuite32_32700.zip)

MrLatebros, 2019-11-02:
No. It fails to write settings as before.

4712, 2019-11-02:
@MrLatebros For me it looks like an internal problem of the Nano board or the connection from Nano to the ESC. Do you have a second Nano to try?

4712, 2020-01-18:
Please check out the new released version 32.7.0.9. It should fix the issue.
It took some time to get the clue about what happens. 
Thanks again for your heads up and the testing!