bitdump/BLHeli #412: Which hardware definition is suitable for the NCP81253 FET driver?

crteensy, 2019-10-29:
The NCP81253 FET driver has a PWM input and an enable input. Application circuit and truth table:

[image]

[image]

Which hardware definition (if any) is suitable for use with this driver?

sskaug, 2019-10-30:
The X_Y_0 code are designed to work with EN/PWM drivers.
Deadtime zero is used to designate EN/PWM.

crteensy, 2019-10-30:
So for example at the top of the "O" hardware definition we find:
"Com fets are active low for H/L_N driver and EN_N/PWM driver"

This would be for an active high high-side FET and an active low low-side FET
OR
for inverted enable and non-inverted PWM. Enable is connected to Ap, Bp, Cp and PWM is connected to Ac, Bc, Cc.

A truth table for O would then be

```
EN_N  PWM  HFET LFET
0     0    off  on
0     1    on   off
1     0    off  off
1     1    off  off
```

Am I getting this right?