bitdump/BLHeli #413: problem with blheli32suite 32.7.0.3

simonefpv, 2019-10-31:
I replaced an esc broken by my build, esc in question tekko32 f3 35a targhet furling32, after the correct installation I immediately updated all it's Four escs, after updating I try the correct functionality but the engine 2 which is the one where I replaced esc runs has traits, I reconnect on blheli32suite and I see that the escs that work have an entry with (bootloader h) instead what does not work (bootloader k) how do I solve?

image is not mine ... but I have the same problem

4712, 2019-11-02:
@simonefpv Different bootloader revisions are in general no problem, do not affect working of the ESC in flight time. What problems do you discover?

simonefpv, 2019-11-03:
nothing I thought was a hardware problem because the old escs turned perfectly but the new one went has clicks ... then I discovered that the new one was defective

closed 2019-11-03 by simonefpv