bitdump/BLHeli #414: problem with blheli configurator via google chrome

DoctorBormental, 2019-11-02:
[Log.txt](https://github.com/bitdump/BLHeli/files/3800484/Log.txt)

I don’t understand what is the problem? After clicking Read setup, nothing happens! Furling 32 flight controller. I connect the battery: then I connect to the computer via usb. then the connect and read setup button, but as a result, the setup menu does not appear? I attach screenshots and log. A huge request for help. Betaflight, by the way, sees the motors.
[image]
[image]

mikeller, 2019-11-02:
@DoctorBormental: You are using the BLHeli configurator (https://github.com/blheli-configurator/blheli-configurator) which is a different project, and does not support BLHeli_32, since it uses a proprietary protocol.

DoctorBormental, 2019-11-02:
Thank you so much for the quick reply! can you advise me how to configure the flight controller in another way?