bitdump/BLHeli #415: blhelisuite refuses to start

hiseob, 2019-11-06:
[image]
[image]
[image]

logged something on event viewer :

`<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
<System>
  <Provider Name="Application Error" /> 
  <EventID Qualifiers="0">1000</EventID> 
  <Level>2</Level> 
  <Task>100</Task> 
  <Keywords>0x80000000000000</Keywords> 
  <TimeCreated SystemTime="2019-11-06T09:58:02.326724600Z" /> 
  <EventRecordID>11621</EventRecordID> 
  <Channel>Application</Channel> 
  <Computer>HISEOB-R7NOTE</Computer> 
  <Security /> 
  </System>
<EventData>
  <Data>BLHeliSuite.exe</Data> 
  <Data>16.7.14.9</Data> 
  <Data>2a425e19</Data> 
  <Data>KERNELBASE.dll</Data> 
  <Data>10.0.18362.449</Data> 
  <Data>a0c7be3c</Data> 
  <Data>0eedfade</Data> 
  <Data>00113572</Data> 
  <Data>2e78</Data> 
  <Data>01d59488ab02dea0</Data> 
  <Data>C:\Users\hiseo\Desktop\BLHeliSuite16714901\BLHeliSuite.exe</Data> 
  <Data>C:\Windows\System32\KERNELBASE.dll</Data> 
  <Data>4cf6852b-0589-4a30-a555-0f1890f16ff4</Data> 
  <Data /> 
  <Data /> 
</EventData>
</Event>
`
(exception code : 0x0eedfade)

cannot get blhelisuite running

downloaded from mediafire, google drive whatever it's not running

completely exhausted with this

should i just throw away all of my escs then buy new stm32 based escs because of this ? ah.h...

tried on fully updated windows 10 pro

4712, 2019-11-06:
@hiseob Did you extract all files of the zip file?

[image]

Are there any more messages from the application?

haslinghuis, 2019-11-06:
AFAICS kernelbase.dll is a generic windows problem. So no single solution. But cannot reproduce on Windows 10 Pro 64bit - just downloaded this build using same method, extracted using 7zip and running without problem. The KernelBase.DLL is a core system file that should NOT be deleted.

BLHeli cannot support Windows problems but a possible solution could be: 
```
Windows 10 has a built in utility called DISM which can be used for health checking and restoration amongst other things.
```
**Please note that I am not responsible for any harm that may befall your system by using the following commands.**
```
Open CMD Prompt as an Administrator (Start > Windows System > CMD > Right Click > Run as Administrator)
Type or Copy and Paste without the quotes - "DISM /online /cleanup-image /scanhealth" This will check your C:\Windows folder for any kind of corruption and inform you if it detects any
If you receive a message along the lines of component store corruption detected run the following command in the same Administrative CMD Prompt - "DISM /online /cleanup-image /restorehealth" - This command will attempt to repair any of the corrupted system files using your recovery partition and I believe Windows Update also.
```

hiseob, 2019-11-11:
@4712 Yes. Extracted all. No message from suite.

@haslinghuis No issues raised from dism check, obviously...

sha256 hashes attached

[sha256sum.txt](https://github.com/bitdump/BLHeli/files/3829517/sha256sum.txt)

I will test on freshly installed english windows 10

hiseob, 2019-11-11:
Running okay with freshly installed and fully updated windows 10 1903 18362.449 on vmware but not my physical machine 

I don't know what's going on...

Anyway, thanks for great job. I will try to use this on vm

anton-matosov, 2020-08-06:
I have experienced same issue today with 16.7.14.9.0.2 on latest Win 10 Home. I have tried many things (admin mode, compatibility modes) but nothing seemed to help. I accidentally let it run in background while I was looking for the next troubleshooting guide and after about 5-10 minutes of no response app finally showed up and it works well ever since.