bitdump/BLHeli #417: Rasor32 défault connection to BLHeliSuite32 32.6.1.6

rghenri, 2019-11-12:
Avec ADK 2650
Make interface bootloader (USB/com)   -> Flash ADK  -> OK

Pin PWM Razor32 on D51 + GND of ADK
ESC setup  /bouton Connect  wait 
On power +24V
Led active 500ms

No connection ?

4712, 2019-11-13:
I cannot reproduce with BLHeliSuite32 32.6.1.6, but with BLHeliSuite32 32.7.0.4.
So, thanks for the bringing this up here!

The Arduino Mega 2650 seems not to like DTR hold high.
This issue is fixed with BLHeliSuite32 32.7.0.5, which is just released.

rghenri, 2019-11-13:
Merci je vais essayer ! Merci ++++

rghenri, 2019-11-13:
Ne marche toujours pas avec Mega ADK 2650, connexion impossible !
Nouvel essais avec Nano pro mini 328. ->OK
Connexion OK readSetup -> OK
Mauvaise   FLASH R_FlyFort rev 32.6. Je cherche le programme pour "Flash Razor32 v2" @+

4712, 2019-11-13:
@rghenri 
BLHeliSuite will normally detect the right ESC hex itself and also contains all available hex files.

Could you please check, if you have selected the right interface type for the Mega?
...And, if possible, please share the log file (log.tlg as zip file).
[image]
[image]

rghenri, 2019-11-14:
Merci pour ces information sur le bon ESC fichier hex, Je vais vous faire parvenir le log dans quelques jours.
je vais raccourcir la longueur de mes 2 x fils (GND & PWM) sur l'ESC et placer des condensateurs plus fort, pour éviter tout phénomène les parasites et recommencer avec un autre ADK MEGA 2650
cordialement
@rghenri