bitdump/BLHeli #418: Bidirectional mode, active braking, non-damped mode, and back EMF

cbixel, 2019-12-03:
I am using Flycolor X-cross 50 amp ESC's in combat robotics.  They seem very promising, but I am working out a few issues.  I am able to use them successfully for the weapon motor by turning braking off, turning "non-damped mode" on, and selecting "Normal" for the motor direction instead of bidirectional.  With these settings, driving the weapon at full power with a 6S battery barely makes the ESC warm at all. 

Any sort of braking or damping absolutely melts the esc. Soldered connections come loose and components fall off.   My conclusion is that the ESC is attempting to slow the weapon, and the amount of energy in a 5 pound chunk of steel is so far in excess of a typical propeller, that the ESC is simply not designed to handle this.  (It could also be that the ESC is attempting to send the energy back to the battery, but the battery is not able to accept that extreme amount of power.  

For the drive motors, the flycolors are working fairly well, but we have had a few of them burn out.  
Unlike the weapon motor, the drive ESC's must be bidirectional.  Would setting "Ramp Up Power" to a lower value decrease the rate at which energy is directed back from the motor?  I guess I am wondering if the Ramp Up Power setting also controls the rate of braking during the slowing down part of a direction reversal.

Also, we are using the xcross 35amp ESC's for the drive motors, and would the 50 amp ones be significantly better at handling the back EMF?

sskaug, 2019-12-03:
Never heard of such issues with braking, but then again props do not have that much inertia.
So apart from setting nondamped mode, there are no settings that control the active braking force.
Rampup power only controls acceleration.

cbixel, 2019-12-03:
Yes, a prop does not really contain much rotational kinetic energy compared to the 5 pound weapon on our bot.  I guess some of the larger helicopters might have a large amount of energy, but they would also have a much larger motor and esc.  The flycolor x-cross are awesomely tiny, which is important for combat robotics.

On another note, as far as bidirectional mode is concerned, I think the rampup power would help with the acceleration part of the direction reversal.  Are you saying that there will still be active braking during the slowing down part of the change of direction?

sskaug, 2019-12-04:
During the slowing down portion of direction reversal, there is actually torque reversal.

cbixel, 2019-12-04:
Is that reverse torque achieved through the same means that braking is achieved?  I would imagine that it is, as this would be the most efficient and would send a bit of charge back to the battery.  It is also possible that it does this by shorting the leads from the motor and thus dissipating the energy in the motor coil.  

For me, the second way would be best, as that would not damage the ESC.  If, however, the reverse torque is achieved by turning the motor into a generator and attempting to send the energy back to the battery, we would have to find some way to handle that amount of energy.

d4rken, 2019-12-23:
> Any sort of braking or damping absolutely melts the esc.

Might be related to https://github.com/bitdump/BLHeli/issues/293