bitdump/BLHeli #420: BLHeli telemetry direct in FrSky receiver

Matheusschmitz, 2019-12-12:
Can I connect the telemetry of ESC direct in a FrSky receiver? I am building a plane without a FC, I want to see the ESC information on the radio screen.

sskaug, 2019-12-12:
Yes - that is exactly the setup we have used during development. Just connect the telemetry pin of the ESC to the S.PORT pin of the receiver. But remember you have to set the S.PORT physical ID using BLHeliSuite32. The telemetry port of the ESC defaults to KISS style telemetry.

Fanfe8, 2020-05-14:
Thanks for the reply! I'm half way there to use ESC telemetry in FrSky receiveir as well for plane or wings. How do you setup the S.PORT physical ID in BLHeliSuite32 ? I only finf the option to set "auto telemetry" on or off. And to know wich is the ID of your S.PORT receiver ?

sskaug, 2020-05-14:
Are you using the Rev32.71 test code?

Fanfe8, 2020-05-14:
Thanks !
I'm on Rev32.7 so that's why I didd't found the sport option. 
The BLHeli32 suite don't allow to select manualy an HEX file for flashing the Rev32.7.1.
How can I flash this firware to the ESC ?
DYS_Aria_32_Multi_32_71.Hex

Saved it as HEX file after opening the code in RAW and put the HEX file in HEXfile folder of BLHeli32 but still not possible to chose more than REV32.7 for flashing.

sskaug, 2020-05-15:
Do you ue the latest BLHeliSuite32?

Fanfe8, 2020-05-15:
I use the 32.7.1.1 version.
Seems to be the latest on the download here:
https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM

4712, 2020-05-15:
@Fanfe8 Depending on the used browser and the OS sometimes something can get wrong when downloading the hex files. Please verify with a text or hex editor it starts like:
> :1020000032B17FF

To check wether it is detected after saved to \BLHeli32_HexFiles\:
Turn on the log function (Options... Enable Log), restart the app and have a look at the log.
It should look like:

[image]

Fanfe8, 2020-05-16:
Thanks ! Something was wrong with the HEX file indeed  (to many space or return key in it).
Now telemetry is working with the DYS ARIA 35A ! Relly nice to see the mah drained during flight with planes or glider ! But dont get the voltage yet (it stay at 0V).

Just tried with another ESC to see if voltage will work. Connected the INI 45A slim ESC and it appears to be named "FLYWOO_GOKU32_PRO" and not "FLYWOO_INI_PRO" as I was guessing.

So not possible to flash it with Rev32.7.1 yet.
Is there a trick to adapt another HEX file for this INI ESC wich is in fact called FLYWOO_GOKU32_PRO ?

sskaug, 2020-05-16:
Great that you now got it working! I've now uploaded the FLYWOO_GOKU32_PRO code

sskaug, 2020-05-16:
Do not try to flash a code with a wrong name!....

sskaug, 2020-05-16:
The DYS Aria 35A does not have voltage sensing hardware. That's why you get zero.
The FLYWOO_GOKU32_PRO should report both voltage and current.

Fanfe8, 2020-05-16:
Appreciate a lot your help ! That's some service !
Was just able to flash the FLYWOO_GOKU32_PRO and got telemetry with voltage and current, yeha ! Current seems a bit off the charts, maybe the current sensor is not working properly. The amps don't rise when pushing the thoottle up.. Will do more testing tomorrow but playing with current adjustment doesnt change much.

gmo78, 2020-12-08:
This discussion was very useful for me. Sadly I discovered my new Tekko32 F3 35A with furling32 firmware is not capable of direct telemetry with frsky (from the BLHeli manual: "If the "S.PORT Physical ID" programming parameter shows up in BLHeliSuite32, then your ESC supports it").  I will try with a kiss esc.

4712, 2020-12-08:
@gmo78 Please use the latest BLHeliSuite32, download the appropriate 32.7.3 test code hex file form [here](https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Rev32.7.3%20Variable%20PWM%20frequency%20testcode) and put it into  BLHeliSuite32\BLHeli32_HexFiles folder and flash.

gmo78, 2020-12-08:
thanks for the answer! I tried also your way, but with no success. It seems the suite did not find any better firmware than the one already installed on it.
[image]
[image]
[image]
[image]
[image]

gmo78, 2020-12-08:
I tried again and... I found the new 3.73 version!
thanks!
[image]

jasc76, 2021-03-31:
What is the correct S.Port ID for direkt connection to FrSky Receiver with OpenTX Radio?
25 ?

gmo78, 2021-03-31:
Mine is working with 25