bitdump/BLHeli #422: 32.7.0.6 write setup failed

yulei1983, 2019-12-18:
**steps**
 1.connect interface
2.read setpup and then power the ESC.(read setup successful)
3.set ESC some values, click write setup.
**expected**:
should be successful!
**Actual:**
Popup message:
---------------------------
Warning
---------------------------
Write Setup section to ESC Failed!
This means: the ESC layout might no longer be identified automatically until next power cycle.
The layout information is:

 HAKRC_35A - Rev. 32.6 - Multi

Please analyze the reason of failure, cycle power and try to reflash the ESC.

---------------------------
Information
---------------------------
Interface does not support Mutiple ESC configuration.

4712, 2019-12-18:
@yulei1983 Could you please provide a full log.tlg file (Log... save to file...) here?

yulei1983, 2019-12-18:
[image]

4712, 2019-12-18:
Perhaps rename log.tlg to log.txt and retry...
BTW: It looks like a driver issue, is it a CH340 chip on the Nano?
You could try to flash the Nano with the Arduino1Wire_Nano_328.hex and try using interface "L".

yulei1983, 2019-12-18:
Yes,CH340

yulei1983, 2019-12-18:
I change music note config ,but popup Nothing to write,and then no success any more. May be caused by this area

4712, 2019-12-18:
Sorry, I cannot reproduce this here. Maybe try another 1wire interface or FC...

4712, 2019-12-18:
BTW,what diver version has windows for the ch340?

yulei1983, 2019-12-18:
> BTW,what diver version has windows for the ch340?

wch.cn
3.3.2011.11

4712, 2019-12-18:
Maybe check out the latest driver at http://www.wch.cn/download/CH341SER_EXE.html
(3.50)

yulei1983, 2019-12-19:
> Maybe check out the latest driver at http://www.wch.cn/download/CH341SER_EXE.html
> (3.50)

I update the COM driver to 3.50,but flash failed.
[image]

yulei1983, 2019-12-19:
> Perhaps rename log.tlg to log.txt and retry...
> BTW: It looks like a driver issue, is it a CH340 chip on the Nano?
> You could try to flash the Nano with the Arduino1Wire_Nano_328.hex and try using interface "L".

CH340G

4712, 2019-12-19:
Thanks for testing the rev. 3.50 driver!
I'm still searching how I can reproduce the failure...
Could you rename log.tlg to log.txt and retry to upload?

4712, 2019-12-19:
All I find until now, points to an USB error, where the interface times out because of missing bytes not received from PC.
Please try to flash the attached hex to the Arduino (unpack to BLHeliSuite32\Interfaces\Arduino4w-if\). 
Timeout for waiting bytes to come in increased from 160ms to 3 seconds. 
[4wArduino_Nano__16_MULTIv20003_Test.zip](https://github.com/bitdump/BLHeli/files/3982752/4wArduino_Nano__16_MULTIv20003_Test.zip)

yulei1983, 2019-12-20:
well,how to use this file to make interface?thx

4712, 2019-12-20:
Thanks, was the log from the new test hex (4wArduino_Nano__16_MULTIv20003_Test.hex), or from the normal one (4wArduino_Nano__16_MULTIv20003.hex)?

4712, 2019-12-21:
Attached is a beta version of BLHeliSuite32.exe (rev 32.7.06.1), which repeats the commands if "I:Unknown Error" occurs.
It may help in your situation, but the final solution should be, to find the USB communication problems.
[BLHeliSuite32.zip](https://github.com/bitdump/BLHeli/files/3991314/BLHeliSuite32.zip)

4712, 2020-01-18:
@yulei1983 Please check out the new released version 32.7.0.9. It should fix the issue.
It took some time to get the clue about what happens. 
Thanks again for your heads up and the testing!