bitdump/BLHeli #423: Incompatible with inrunner?

cluhh, 2020-01-04:
I tried a DYS ARIA 70A ESC with a 820kv inrunner(KB 4474) and it has an issue where the motor doesn't go above about 5% throttle.  Any throttle above 5% acts the same as if it was at 5% throttle.  Below 5% throttle seems to be running at 5% as well, but with periodic stutters, increasing in frequency as the throttle approaches 0%.

I am running BLHeli_32 Revision: 32.7
I tested with both 6s and 3s, and both act the same, except that 3s is half the speed.
I tested this with 2 different motors and 2 speed controls of the same type to confirm that the behavior is consistent.
The motor works as expected with another ESC I have(40ASBEC).
The DYS ARIA 70A ESC works as expected with a different motor I have (Turnigy D2826/6).
I tried adjusting Motor Timing and Demag compensation, which seem to have no effect.

Are there any settings I can change that would potentially fix this issue?  Is any additional information needed to debug this?

Thanks!

sskaug, 2020-01-04:
This most likely has to do with low rpm power protection. You can try to turn it off. Or, if increasing rampup power helps, then that is better than turning off low rpm power protection. If you get problems with desyncs, you may have to set current protection to some limit.

cluhh, 2020-01-04:
Adjusting rampup power didn't fix it, but disabling Low RPM Power Protect worked!  Thanks for the help!