bitdump/BLHeli #424: Error: Found 2 ESC with similar unique ID numbers

SandroFTW, 2020-01-04:
I'm trying to use 2 different 4in1 BLHeli_32 ESCs to build a 10" Hex. Both of them only have 3 working outputs after having been in use in racing quads for a while (Hobbywing XRotor 45A + iFlight SucceX 50A).

I tried iNav and Betaflight as flight controller firmware and DShot600, Multishot and Oneshot42 as protocols (also did the calibration process multiple times).

When using the Master motor slider in the FC GUI all motors spun up normally but the individual sliders behaved weird:

1: Motor 1 and 5 start twitching
2: Motor 2 and 6 start twitching
3: Motor 3 turns normally
4: Motor 4 turns normally
5: Motor 1 and 5 spin but stutter from time to time (basically exactly the opposite of Slider 1)
6: Motor 2 and 6 spin but stutter from time to time (basically exactly the opposite of Slider 2)

When I try to connect to BLHeli Suite I get "Found 2 ESC with similar unique ID numbers!" as an error message and clicking OK closes the software. Disconnecting either of the 4in1 ESCs and swapping around the motor signal connections on the FC doesn't change the behavior.

[image]

4712, 2020-01-04:
@SandroFTW Most likely there is a short between two of the ESC signal inputs.

SandroFTW, 2020-01-05:
Checked visually and with a multimeter. I can't find any shorts.

4712, 2020-01-05:
Two ESCs might be hardware or software wise connected to the same motor output of the FC.

SandroFTW, 2020-01-05:
That would be my only explanation. Motor 5 and 6 are colored differently on the pinout diagram.

P.S. On DShot the motors don't stutter but increasing Output one spins motor 1 and 5 normally. Outputs 5 and 6 do nothing. (That confirms the theory)

SandroFTW, 2020-01-05:
I used a different FC. Apparently this one doesn't like to be used with 6 ESCs. The new one works flawlessly. I will transfer the issue to the FC software devs.

closed 2020-01-05 by SandroFTW

4712, 2020-01-05:
Thanks for the info!