bitdump/BLHeli #425: BLHeli32 - 32708 - CH340 based interfaces fail to write / flash

airzone-sama, 2020-01-12:
Basically the same as issue 422 and 411.

Interfaces using CH340 USB bridge often fail to save configuration, always fail to flash. I've tried a random selection of approx 80 or so atmega328p w/ ch340 boards I have, plus a USB linker based on this chip. For the atmega based boards, I've tried 1-wire and 4-wire interfaces.

Interestingly, Arduino boards with an atmega16u2 USB bridge work fine.

CH340 Drivers tested: 3.4.2014.8 and 3.5.2019.1. I have also used 2 different computers, both Win 10 and both the same result.

Attached log file showing a failed flash.

[Log.tlg.txt](https://github.com/bitdump/BLHeli/files/4050091/Log.tlg.txt)

4712, 2020-01-12:
@airzone-sama Thanks a lot for your great effort!
Very interesting is the fact that reading the echo bytes failed always after 160 bytes.
I could not reproduce the failure here with my win 10 system, may be it is related to the main board USB interface.
I will try to investigate further and for sure come back with test versions...

4712, 2020-01-13:
@airzone-sama WOuld be great, if you could help with a small test.
I made a small test app to check the 1-wire if the write and read over USB works.
Simply connect the Arduino (no ESC connected), start the app, select the right com port and click "Send bytes A". If it fails check out "Send bytes B".

[image]
[Test1.zip](https://github.com/bitdump/BLHeli/files/4055583/Test1.zip)

airzone-sama, 2020-01-15:
OK thanks. It's been a hectic few days, so I haven't been able to do further testing.

4712, 2020-01-18:
@airzone-sama Please check out the new released version 32.7.0.9. It should fix the issue.
It took some time to get the clue about what happens. It turned out that setting the WriteTotalTimeoutConstant while a current asynchronous write is pending was the problem. Weird, that the issue does not happen with FTDI devices. And for CH340 (and maybe other brand USB/UART chips) it does only seem to happen with specific USB I/O chip sets. Maybe it is also dependent of the OS version and USB driver installed. 
Now  I switched to a fixed write timeout and to waiting for asynchronous write operation to finish before start reading. 

Thansk again for your heads up and the testing!