bitdump/BLHeli #427: (re)Flashing bootloader on Racerstar PG30 30A ESCs

s-ol, 2020-01-27:
I'm having trouble with Racerstar PG30 ESCs. I bought four new ones, and only two are being read and flashable via BLHeliSuite32xl (via 4way FC). Based on some posts I'm assuming I need to reflash the bootloader on the two disfunctional ones.

There are four pads on the back of the PCB, but I'm not sure how to tell which is which and used for what. I peeled a sticker off the MCU on one of them, it reads `SPIN32F0 / FMDIV A6 / THA IF 647`, so it seems its an ST *Advanced BLDC controller with embedded STM32 MCU*. The datasheet is available here: https://www.st.com/resource/en/datasheet/stspin32f0.pdf

I'm looking at the pin table but don't see the debug pins mentioned clearly. Would be very glad if someone could point me in the right direction.

s-ol, 2020-01-27:
If the orientation is based on the text on the IC, then the first two pads are `OP4P` and `OP4N` (pins 1 and 2) respectively, the next is either `VM`, `SW`, or both (pins 8 and 9), the last seems to be `PB1` (pin 19). There is one more test point but it doesn't seem to be connected to the MCU directly.

I'm finding it really hard to take a picture of this, but here's my best attempt so far:
[image]

s-ol, 2020-01-27:
Just noticed that there does appear to be a dot marking the corner closest to test point 3, it's a bit hard to tell with all the glue residue. If that is the pin-1 corner, then the TP-pin-mapping would be

| TP | pin no  | label        |
|===|======|========|
|1    | 37 | PA13_SWD_IO |
|2    | 38 | PA13_SWD_CLK |
|3    | 44 (?) | GND |
|4    | 7 | NRST |

... that looks a lot better and more realistic. SWD is the system I2C debug interface, so I'm guessing I'll have to wire up that and ground to get talking to it.

s-ol, 2020-01-27:
The only question now is which of the 4way-arduino mode to use, and which pins?

sskaug, 2020-01-27:
Unfortunately, for BLHeli_32 the bootloader can not be updated.

s-ol, 2020-01-27:
@sskaug: why, because is it fused or something? If it is fused, then there is only two options:
- it was never flashed at the factory, and thats why my ESCs are broken (so I can still do that)
- it's still intact and something else is broken, in which case that other thing is the problem.

s-ol, 2020-01-28:
I've desoldered the two malfunctioning ESCs, and when I power them they do the three tone startup sequence, however I still cannot connect to them. I flashed an Arduino Uno as a 4way interface and I can talk to the other two individually fine (GND and Pin 11 connected), but the two ESCs in question still don't show up. I de- and resoldered the signal wires and tried it both with signal and battery ground connected as well.

s-ol, 2020-01-28:
In the end it turns out the ESCs were fine and stuck in PWM mode, but the PDB I used had a short and the two signal channels were locked together.

closed 2020-01-28 by s-ol