bitdump/BLHeli #428: BLHeli_S firmware questions

HarjitSi, 2020-02-01:
I've read the BLHeli_S firmware a few times and had some questions that I'm hoping you all can help with.

Questions about BLHeli_S with **A.asm**. I arbitrarily chose **A.asm**.

**A.asm:**
Hardware definition file "A"
; X X RC X MC MB MA CC X X Cc Cp Bc Bp Ac Ap

1) What does RC mean?

2) Are the following correct?
MC: Motor C phase connection
MB: Motor B phase connection
MA: Motor A phase connection
CC: Motor A, B, C each connected to a resistor and the other side of the resistors are connected together and this is the node?
Ac: Motor A phase half bridge high side FET. This is driven high or low only.
Ap: Motor A phase half bridge low side FET. This is PWM
Bc: Motor B phase half bridge high side FET. This is driven high or low only.
Bp: Motor B phase half bridge low side FET. This is PWM
Cc: Motor C phase half bridge high side FET. This is driven high or low only.
Cp: Motor C phase half bridge low side FET. This is PWM

3) Does: [ABC]comFET_off mean turn off the high side [ABC] half bridge FET?

4) What does the "com" stand for?

5) What do the following do:
AcomFET_off
AcomFET_on
ApwmFET_off
ApwmFET_on
Set_PWM_A

I can't tell if this is software PWM or if somehow the PCA (or PWM) is being routed through the crossbar and it is being output on a pin or?

6) In BlHeli_s.asm, init_start_bidir_done:
Why are comm5comm6 and comm6comm1 called back to back?

7) I'm struggling a bit with the calc_next_comm_timing: routine.
It looks like the comparator is not generating an interrupt but is polled. Is that correct?

8) For spin up, when there is no BEMF to generate a comparator transition, the code relies on the commutation period timing out?
I haven't been able to work out how it reduces the time per commutation phase so that the motor accelerates. Is this done in the code just before setup_zc_scan_timeout_startup_done:?

I didn't see how it terminated / transitioned from startup to using BEMF on acceleration and then on deceleration from using the comparator to just the timer

9) Is the basic logic - use timer 3 to create the window in which the comparator for BEMF zero crossing should occur and if it does, use that timing to setup the next commutation cycle and if it doesn't, then ?

Thanks in advance.

HarjitSi, 2020-02-10:
Help?

crteensy, 2020-02-24:
I've been trying to find answers to many of these questions as well. Here's what I can say:
1. RC is the receiver signal (I'm not sure if anything other than DShot is supported as of now, earlier versions might have supported more protocols). For most pinouts, P0.5 coincides with UART0_RX.

2. MA, MB, MC, CC: Correct. You need voltage dividers between the motor pads and gnd. The centers of these go to MA, MB and MC. Another resistor from these goes to the center node CC.
Ac, Ap, ... are the "gate" outputs. I'm not sure if Ac is always high side and Ap is always low side. Same for the PWM role. My impression (!) is that this is not always the same across the different pinouts, but I may well be wrong. "O" pinout, for example, is described as "Like D, but low side pwm". It's confusing.

3. the code implies that the [ABC]comFET_off macros are indeed used to turn of the com FETs. However, I'm not sure if some crossbar trickery delays that until some timer event occurs.

4. I'm torn between "com"plemetary and "com"mutation. So yeah, I'm clueless.

5. See 3 for the com/pwm_on/off macros. Set_PWM_A and siblings are more complex because they also manipulate P1SKIP, which is used for crossbar functionality. I'm trying to figure out exactly that for my own purposes at the moment.

Can't help with the others (yet, hopefully).

wjxway, 2024-03-21:
The lack of hardware design guide for BLHeli or similar ESC firmwares is really disappointing. It took me a few days running through scattered tutorials and the .inc files to understand how could I design one on my own. 
BTW, I believe Ap and Ac are A_PWM and A_Common respectively, the PWM pin will pulse to create the desired PWM value, while the Common pin will stay high continuously when A phase should be connected. You could use either high-side or low-side PWM, I don't think there's much of a difference, but most use low-side PWM.