bitdump/BLHeli #429: BLHeli_32 ESC Telemetry won't read over 42k rpm

MrLAtt, 2020-02-03:
As title, I am using a Tekko 32 metal 65a ESC with a 2450kv 14pole motor on 8s, the RPM telemetry reads fine up to 42,000 rpm, but after that the teletry reading just stays at 42k, regardless of increasing motor rpm.

Is this a known issue or limitation? or is it a bug? It would be very helpful if this worked, because it only reading up to 42k is unfortunately of no use to me.

sskaug, 2020-02-03:
This is strange. There is no such limit in the BLHeli_32 code nor in the dataformat for erpm.
I just tested here, with an F3 ESC, running a 4000kv motor on 4S, and it goes up to 370k erpm. Which would be almost 53000 rpm with a 14 pole motor.

MrLAtt, 2020-02-04:
Interesting, it is definitely happening, is there any other information that would help with diagnosing the issue?

I forgot to mention before, I am using Inav 2.3, and I am only trying to display the rpm on the OSD. It seems unlikely though that that's the issue, since it's just displaying the number.

Should I ask in the inav GitHub?

Could it be my ESC? I don't really see how, because if it didn't know the rpm, it wouldn't be able to drive the motor. It's very disappointing, because my motor is spinning above 55k rpm in flight.

sskaug, 2020-02-04:
I have no knowledge of Inav. You may try with another system (BF FC, S.PORT RX etc). And you may ask Inav devs.

MrLAtt, 2020-02-05:
Thank you for the help!
I just confirmed with the INAV guys that it's an overflow error on the average rpm calculation in INAV. It will be fixed in the next release.

closed 2020-02-05 by MrLAtt