bitdump/BLHeli #432: BLHeli32_Aikon_AK32_4S_V1_0 - Rev. 32.7 - one side of Bidirectional is inverted

jimkazmer, 2020-02-15:
Hi,  You can see my BLHELI_32 config info via the attachment below. I am using dShot600 to communicate with one ESC. I am cycling through speeds starting at 1024 (stopped) and going up to 2040 (my max forward speed) then back down to 1024... all that works great.  Once I send the 1023 speed command, the motor jumps to max reverse and as I step downward to 50 it slows down, then as it cycles back up to 1023 it speeds up to max reverse. Once it hits 1024 it stops and the progression repeats.

What is causing this?  I have used this ESC and motor combination just fine with OneShot125 and it did not behave this way.  I need to switch to dShot for many reasons.

Any insights?  I did try three of the four bidirectional modes and they all behaved the same.

[BLHeli32_Aikon_AK32_4S_V1_0 - Rev. 32.7 - Multi_200214.txt](https://github.com/bitdump/BLHeli/files/4207700/BLHeli32_Aikon_AK32_4S_V1_0.-.Rev.32.7.-.Multi_200214.txt)

sskaug, 2020-02-15:
Yes. This is the way bidirectional operation is specified (by KISS) for dshot.

jimkazmer, 2020-02-15:
I am a bit surprised it played out that way.  But looking back on my past designs, I know I've done similar things.  Thank you for getting back to me so quickly.

closed 2020-02-15 by jimkazmer