bitdump/BLHeli #434: CMD 33

jimkazmer, 2020-02-17:
I have dShot 600 &1200 working with a teensy microcontroller.  I can request telemetry and read the results just fine.  I am using BLHELI_32 version 32.7.

I'd like to us CMD 33: 
DIGITAL_CMD_SIGNAL_LINE_CONTINUOUS_ERPM_TELEMETRY
which I assumes outputs the eRPM in a continuous manner (very frequently).  The manual says: 
33 DIGITAL_CMD_SIGNAL_LINE_CONTINUOUS_ERPM_TELEMETRY // Need 6x, no wait required
Commands are only executed when motors are stopped.

Does BLHELI_32 support CMD=33?  If so, can you describe more of the requirements for to activate this command?  
Does it need to be re-issued each time I stop (CMD=0)?
How many times do I need to send it, and how long between messages?
Do I need to set the telemetry bit when I send this command?
Do I need to follow the command with one or more stop commands?
If I do get CMD33 working, how frequent are the updates?
Is it the entire telemetry message or just the two ERPM bytes with or without a CRC?

Sorry for all of these questions.  I am really psyched to be close to device testing (as opposed to bench testing).

mikeller, 2020-02-17:
@jimkazmer: It looks like the documentation that you are using is outdated. Dshot RPM feedback switched to a different protocol that uses an inverted signal with uninverted checksum to signal to the ESC that RPM feedback is requested. For details of the implementation see betaflight/betaflight#8146.

jimkazmer, 2020-02-17:
Thank you for that fast response and useful information.
Do you know if the eRPM telemetry is still output on the telemetry wire?  I feel like I may still be missing something... is the wire that sends the Dshot commands now a bi-directional communication wire?

The root cause for doing "inverted Dshot" was experienced during my hardware implementation, and I actually bought hardware inverters to convert the pulled-up high IDLE to a low IDLE.  Switching to "inverted Dshot" will allow me to remove a component (switching back to the logic level converter) and I'm done.  Implementing a bi-directional communication line will be a bit of (software) work.  If the bi-directional communication wire is real, can you point me at more details about its implementation (what I'd need is timing info... how long between do I wait for the eRPM telemetry, is it inverted as well, is there a CRC).

mikeller, 2020-02-17:
@jimkazmer:

> Do you know if the eRPM telemetry is still output on the telemetry wire? 

No, it never was - it is sent on the Dshot signal wire.

betaflight/betaflight#8146 has the details of the new RPM feedback protocol that was adopted by BLHeli_32 as receiver side implementation in the code, and the specification in the discussion.

jimkazmer, 2020-02-17:
Thank you... I read #8146 (twice)... I'll look at the code.

sskaug, 2020-02-17:
Maybe you have looked at an older version of the spec. Commands above 29 were removed at some point during bidir dshot development: https://github.com/bitdump/BLHeli/blob/master/BLHeli_32%20ARM/BLHeli_32%20Firmware%20specs/Digital_Cmd_Spec.txt

jimkazmer, 2020-02-17:
Thank you for that; I'm back on track.  Closing this Issue.

closed 2020-02-17 by jimkazmer