bitdump/BLHeli #435: Is anything wrong with this telemetry picture?

jimkazmer, 2020-02-18:
I have "inverted Dshot" working; I can cycle the motor up and down. After each command, I stop driving the wire and wait for telemetry.  This is what I see on the scope:
[image]
On the left side you can see the end of my dShot speed command, and in the middle you can see the telemetry response.

This was taken close to when my motor was spinning at max speed (1000Kv at 12 volts).  The telemetry response is very consistent ranging from 30usec to 32usec after my speed command. However, there are only seven bits in the telemetry response, I expected 16 (plus a CRC).  Also, when I measure the bit timings they vary from bit to bit and range from 3.48usec and 6usec, which is a data rate less than 300kbps.  The behavior didn't change as I switched between Dshot600 and Dshot1200.

This doesn't seem right... I feel like I am missing something. 

I am using BLHELI_32 v32.7.  My update rate for Dshot is 1000Hz.
[BLHeli32_Aikon_AK32_4S_V1_0 - Rev. 32.7 - Multi_200214.txt](https://github.com/bitdump/BLHeli/files/4217492/BLHeli32_Aikon_AK32_4S_V1_0.-.Rev.32.7.-.Multi_200214.txt)

mikeller, 2020-02-18:
@jimkazmer: This looks like a valid RPM feedback response, compare to https://github.com/betaflight/betaflight/pull/8554#issuecomment-512363965

Keep in mind that, unlike the Dshot signal itself, the RPM feedback is GCR encoded, so it does not have 2 transitions per bit.

jimkazmer, 2020-02-18:
That is incredibly helpful information.  I did not know that.  Thank you.

closed 2020-02-18 by jimkazmer