bitdump/BLHeli #440: Changing Dshot specifications and updating documentation

jimkazmer, 2020-02-26:
I'm very excited and impressed with what the developers have done with bidirectional Dshot.  Thank you for this work.   You likely already have thought about the issue below, I am just not able to find those discussions.

The BLHeli_32 manual (page 8) shows these "Maximum supported speeds:"
- Dshot at 8kHz: 470k erpm
- Dshot at 16kHz: 420k erpm
- Dshot at 32kHz: 310k erpm

However, based on my calculations (see below), since bidirectional Dshot requires a 30 microsecond gap between the Dshot command and the returned telemetry, Dshot600 can only support 11KHz and Dshot1200 can only support 16KHz.

Are you planning on decreasing the 30 microsecond delay, and scale it with the Dshot rate?  As you know (or can see below) the other two data "windows" scale with the rate.

Hz | micro-sec between requests
-- | --
8000 | 125.0
12000 | 83.3
16000 | 62.5
24000 | 41.7
32000 | 31.3

Dshot     rate | usec/bit | 16 bits | gap | telem | total
-- | -- | -- | -- | -- | --
600 | 1.67 | 26.67 | 30.00 | 28.00 | 84.67
1200 | 0.83 | 13.33 | 30.00 | 14.00 | 57.33
2400 | 0.42 | 6.67 | 30.00 | 7.00 | 43.67

With the availability of 72MHz and 108MHz ESCs, there likely doesn't need to be a delay of 30 microseconds at these higher speeds.

sskaug, 2020-02-29:
The rpm limits you quote are without bidirectional dshot.

As for the 30us delay, many ESCs can do the turnaround faster. The 30us is also set to give the FC time to do the tx/rx turnaround.

jimkazmer, 2020-02-29:
That makes sense about those values **not** being for bidirectional dshot... was that written before bidirectional dshot was defined?  Won't the majority of users of dshot want telemetry (and therefore be using bidirectional dshot)?

I had not thought about the time needed by the FC to do the tx/rx turnaround.  Is that same amount of time needed on the other side (to change from Rx mode back to Tx mode)?  If it is, then dshot600 is limited to 11khz and dshot1200 would be limited to ~11khz (see tables below).

Thinking longer term, these speeds can easily be increased. The turnaround time on my board to swap between Rx/Tx is less than one microsecond.  I think scaling the delay for the Rx/Tx swap with the data rate (on both sides) makes a lot of sense.  See the third table below for the rates that would be achievable.

Dshot rate | usec/bit | 16 bits | FC Tx/Rxturnaround | telem | FC Rx/Txturnaround | total
-- | -- | -- | -- | -- | -- | --
600 | 1.67 | 26.67 | 30 | 28 | 30 | 114.67
1200 | 0.83 | 13.33 | 30 | 14 | 30 | 87.33
2400 | 0.42 | 6.67 | 30 | 7 | 30 | 73.67

Hz | usec betweenrequests
-- | --
8000 | 125.0
9000 | 111.1
10000 | 100.0
11000 | 90.9
12000 | 83.3
13000 | 76.9
14000 | 71.4
15000 | 66.7
16000 | 62.5
24000 | 41.7
32000 | 31.3
40000 | 25.0
48000 | 20.8
56000 | 17.9
64000 | 15.6
72000 | 13.9
80000 | 12.5
88000 | 11.4
96000 | 10.4
104000 | 9.6

Dshot rate | usec/bit | 16 bits | FC Tx/Rxturnaround | telem | FC Rx/Txturnaround | total
-- | -- | -- | -- | -- | -- | --
600 | 1.67 | 26.67 | 30 | 28 | 30 | 114.67
1200 | 0.83 | 13.33 | 15 | 14 | 15 | 57.33
2400 | 0.42 | 6.67 | 7.5 | 7 | 7.5 | 28.67
4800 | 0.21 | 3.335 | 3.75 | 3.5 | 3.75 | 14.335
9600 | 0.105 | 1.6675 | 1.875 | 1.75 | 1.875 | 7.1675

mikeller, 2020-02-29:
@jimkazmer: Maybe it would be helpful if you stated what goal you are trying to achieve with this discussion / any proposed changes.
As has been shown in a lot of practical testing, there are no gains to be had in increasing the motor update frequency beyond what is already possible with Dshot600 (or in most cases even Dshot300), and in fact support for Dshot1200 has been dropped by most flight controller firmwares as the problems that it created (high error rate due to hardware limitations) outweighed the marginal (at best) benefits.

jimkazmer, 2020-02-29:
I wasn't aware that support for Dshot1200 was being dropped by most flight controller firmwares.  I am having great success reading in telemetry with Dshot1200, and was hoping to go faster than 10khz. To get the telemetry error rate down to 1 (or less) in 100,000 I had to be careful with my wiring, but that just suggests proper shielding or tighter twisted pairs are required.

My first goal was to make sure I understood how things work with the technologies I am trying to leverage.  My second goal was to share my insight about a possible improvement... scaling the Rx/Tx turnaround times (both sides) with the data rate instead of having it fixed at 30usec (there is a symmetry to it).

Faster ESCs (108 Mhz) and FCs (600 Mhz) are here, and even faster ones will come. The improvements you're making with ESCs and FCs will lead to better motors that will be capable of sensing a slight shift in the wind, if not react to it.

I guess you would view what I am doing as experimental, and you may be right that it may have no practical benefits.  My current interest is in getting better eRPM telemetry data.  Adjusting motor speed at high frequencies, although very much desired, is limited by the current motors available. 
 My first step is to test and measure the limits of my components, starting with ESC and motor response times.

mikeller, 2020-03-01:
@jimkazmer: Fair enough that you are testing the limits.
However I think a lot of the limitations around improvements possible through faster motor updates stem from the physical limits of motors, and of how fast they can be accelerated / decelerated. There probably are some improvements to be had, but I think with current motors we are very much into diminishing returns already, especially if cost is to be factored in.
With respect to scaling the switchover times with protocol speed, I don't think this makes sense - there is a fixed amount of time that is needed to switch pins between input and output, and reconfigure DMA, and so on. Plus the protocol should allow some leeway for different implementations / hardware types to facilitate adoption of the protocol.
This is probably less of a limiting factor on the ESC side where only one pin needs to be switched, but on the flight controller side things get considerably more difficult, with multiple pins having to be switched, and other tasks (like gyro readings) having to be run with strict timing constraints.

jimkazmer, 2020-03-01:
Well, hopefully I will get somewhere with what I'm doing.  If I find anything interesting I put it in a post.
Thank you for your responses, and insights.

closed 2020-03-01 by jimkazmer