bitdump/BLHeli #443: Forward power is twice Reverse power

jimkazmer, 2020-03-08:
I am seeing half the RPMs in "reverse" than I am in "forward".  I am using:
- BLHELI_32 v32.7. 
- Update rate for bidirectional Dshot is 8000Hz.
- [BLHeli32_Aikon_AK32_4S_V1_0 - Rev. 32.7 - Multi_200214.txt](https://github.com/bitdump/BLHeli/files/4302656/BLHeli32_Aikon_AK32_4S_V1_0.-.Rev.32.7.-.Multi_200214.txt)
- I have two motors/ESCs and both are behaving the same way.
- I am reading the returned telemetry, and it shows

**FORWARD SpeedCmd= 2047**
- Right  Motor RPS=192.45  Motor RPM=11547.19  Avg RPM=11594.53
- LEFT   Motor RPS=193.35  Motor RPM=11600.86  Avg RPM=11600.86

**REVERSE SpeedCmd= 1047**
- Right Motor RPS= 98.73  Motor RPM= 5924.06  Avg RPM=5922.42
-  LEFT Motor RPS=100.16  Motor RPM= 6009.61  Avg RPM=6009.61

Any ideas why it would be this way?

jimkazmer, 2020-03-08:
I tried swapping two (of the three) motor power wires that come out of the ESC and had the same results.

sskaug, 2020-03-08:
[image]
[image]

jimkazmer, 2020-03-09:
Okay... I went out and bought a flight controller, installed betaFlight configurator, upgraded the firmware to the latest BetaFlight version, and was able to reproduce results similar to yours using that system... How do you specify an 8000Hz Dshot update rate via that tool?  I can't tell what the update rate is. 

I can see the (inverted) Dshot values being sent to the ESCs via the scope (both 1047 and 2047).  I can see the telemetry values being returned are 2x more in one direction than the other. My hand held tachometer also verifies a 2x difference. (and my ears can hear a difference as well).  Something is causing the ESC to perform differently...

What information can I provide, and attain that will help?

jimkazmer, 2020-03-09:
I found my problem. I regret wasting your time, yet your demonstration on how to test this did lead me to discover my issue.  Thank you.

At some point, as I was re-factoring my code, which occurred after my initial verification on the scope of my bidirectional dshot signals, I removed the telemetry bit (since I was no longer using it).    This, of course, resulted in value that was half the intended value. I misinterpreted the my scope signals last night; this morning I decided to look again.

Once again, thank you for your assistance.

closed 2020-03-09 by jimkazmer