bitdump/BLHeli #445: Tekko32 F3 4in1 40A, initialization problem

melenefoti, 2020-03-12:
Dear devs hello,
   I have a Tekko32 F3 4in1 40A ESC, i have installed the latest firmware TEKKO32_F3_4in1 Rec 32.7.
   FC Matek F722 SE (also latest firmware of BetaFlight)
   I am using rpm filtering (DSHOT300 4k/4k). 
   I followed all the necessary setup and settings needed as mentioned on BetaFlight Wiki and youtube videos.
   I have added also a capacitor on ESC battery pads.
  
   Sometimes, when i arm the quad, the ESC initialization tones are not synced (sometimes 1-2 seconds later some motors than the other motors) .
   Other times, motor 3 and 4 dont spin when i arm. I unplug the battery and plug again and arms properly and flies without any problems.
   Most of the times, sounds are all synced and when arming all motors spin.
   Do you have any suggestions? Is there something you might need to check? Or somethings i am doing wrong?
Thank you very much

I also include the CLI command and values specifically for DSHOT.
CLI 
set dshot_idle_value = 550
set dshot_burst = OFF
set dshot_bidir = ON
set dshot_bitbang = AUTO
set dshot_bitbang_timer = AUTO
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set motor_pwm_inversion = OFF
set motor_poles = 14

sskaug, 2020-03-15:
Please try the testcode here: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Rev32.7.1%20SBUS%20and%20S.PORT%20testcode
We implemented a fix for out of sync initialization in this code.

melenefoti, 2020-03-15:
Thank you for your reply. I installed this firmware.
I power up and unplug the battery 10 times. Each time i wait almost 2 minutes for let ESC capacitor discharge totally. 
The motor tones are synced most of the times. Just 1 time (out of 10) one motor was not synced, but the time diff was much less (less than 0.5 secs). 
This firmware shows to be much better than the previous one, the previous one showed unsync 3 out 10 times (with same process) and the unsync was 1-1.5 secs. 
I will use that firmware (32.71). Its getting better.
Thank you very much devs.

melenefoti, 2020-03-16:
I will close this issue as 90-95% solved. 
Thanks for your help.

closed 2020-03-16 by melenefoti