bitdump/BLHeli #446: Questions about Dshot telemetry data (see graph) and sampling

jimkazmer, 2020-03-18:
The graph below shows the telemetry output on the vertical axis (converted to Rotations Per Second) for two motors as the motors accelerate from 400 to 500 using bidirectional Dshot-600.  The horizontal axis is time; the first data point on the left is just before the ESCs receive the Dshot speed command for 500;  The last data point on the right is 100ms later.  In this test, the motors are independent of each other and have a very small load (a 9.25 gram wheel spinning in air). The ESCs are being commanded/sampled at 8000Hz. The motors are 1050kv; the battery is a 3s.

The data shown is not being filtered/averaged in any way by my program; the ESCs may be filtering it, but what is shown is what was received (converted to RPS).

[image]

Questions

- You can see the stair case; its mostly upward, but has many downward steps.  Is there an explanation for these downward steps?  Can they be eliminated?
- What can be done to make the system have tighter/smaller variations?
- As the graph shows, the same telemetry samples are repeated; mostly in runs of 13 and 14, occasionally fewer, with the smallest run being 6 (found once).  Are there any settings in BLHeli configuration that impact the telemetry sample rate that I can adjust/impact?
- If I switch to Dshot-1200 or Dshot-300, would the sampling behavior change? (I plan to test this but thought I'd ask.)

Even when I have both wheels linked (via ground contact), I get similar variations as shown above.  This makes me think the variation is mostly a measurement thing (as opposed to the wheels actually fluctuating like that in realtime -- it seems unlikely the wheel speed changes ~5RPS in a 1.4ms interval).

I plan to do the following additional tests, yet the additional testing I do would benefit from your insights:
* Shorter Wires vs Longer Wires
* Compare Capacitor Types/Sizes, and test with no capacitor.
* Smaller battery vs. larger battery.

sskaug, 2020-03-18:
Without knowing your data in detail, I would say that this is most likely the inherent jitter in BLHeli. Inherent jitter is high, because that is beneficial for smooth running. RPM data is updated once per electrical rpm, so if the dshot rate is faster than that, then measurement values will be repeated. If you need more accurate rpm data, then filtering is the way to go. But this will of course introduce latency, so it is not done in the ESC, but rather left to the preferences of the application.

jimkazmer, 2020-03-18:
Now that you mention that sampling occurs "once per electrical rpm" I do recall reading that in the documentation.  That is very good for me to be mindful of since it suggests how a "correct" sampling/smoothing algorithm should use this data.  Even without that information, I am getting very reliable telemetry results with my averaging/smoothing algorithm.

I'll post the results of my tests if they in any way reduce the "inherent jitter".