bitdump/BLHeli #447: Building BLHeli_S without purchasing Simplicity Studio

okalachev, 2020-03-20:
I barely managed to compile it on MacOS with free version of Simplicity Studio. That's what I get now:

```
*** ERROR L250: CODE SIZE LIMIT IN RESTRICTED VERSION EXCEEDED
    LIMIT:   0800H BYTES
```

How it's possible to fix that?

mikeller, 2020-03-20:
@okalachev: What you need to do is to start Simplicity Studio, open the project in it, and build it - there will be an error message, and a popup with a link allowing you to register for a free license of Keil, and instructions on how to install the license. Once you are able to build the project in Simplicity Studio, building with make will work as well.

okalachev, 2020-03-22:
Hi! Thanks for the help. But I don't completely get, how should open BLHeli project in Simplicity since I cannot find any project-files in this repo.

I tried to create a new project on Keli 8051 platform, and then I received and submitted to Simplicity the free registration key, and compiled my project. But I still get the same error when I try to compile `BLHeli_S SiLabs` with Makefile

mikeller, 2020-03-22:
@okalachev:

> I tried to create a new project on Keli 8051 platform, and then I received and submitted to Simplicity the free registration key, and compiled my project. But I still get the same error when I try to compile BLHeli_S SiLabs with Makefile

Ok, then I can not help you much more - this worked for me.

okalachev, 2020-03-22:
@mikeller, is it possible to open `BLHeli_S SiLabs` as a Simplicity Studio project?

mikeller, 2020-03-22:
@okalachev: I think building it in Simplicity Studio is possible, yes.

okalachev, 2020-03-23:
But is there a ready project for this?

mikeller, 2020-03-23:
@okalachev: No.

thijses, 2021-05-21:
this worked for me (but it took me about 4-5 hours to get to this point, so maybe this information should be noted in the readme file? (for other idiots like me))