bitdump/BLHeli #448: 32.7.1.1?? At least show what has been updated.

AntDX316, 2020-03-21:
When the second set of numbers changes then firmware update?  You don't have to update if the 4th set of numbers are the only change?

4712, 2020-03-21:
@AntDX316 No, **32.7**.1.1 shows that the Suite is capable of working with BLHeli32 rev. **32.7** firmware, the next following numbers from .0.0 to (theoretical) 9.9 are only related to the BLHeliSuite32 versions and not related to the firmware. For info about the detailed changes please have a look to BLHeliSuite32 history.

AntDX316, 2020-03-21:
> @AntDX316 No, **32.7**.1.1 shows that the Suite is capable of working with BLHeli32 rev. **32.7** firmware, the next following numbers from .0.0 to (theoretical) 9.9 are only related to the [...]

Has anything been added?  Why even update to x.x.x.1 from x.x.x.0 ????

AntDX316, 2020-03-21:
Are there notes to see what has been changed?

4712, 2020-03-21:
Either
BLHeliSuite32_327xx\Manuals\BLHeliSuite32History.pdf
or
[image]

AntDX316, 2020-03-21:
> Either
> BLHeliSuite32_327xx\Manuals\BLHeliSuite32History.pdf
> or
> [image]

How come the history local PDF doesn't show up at the place where you are going to download the file?  It's better than downloading it and then checking.