bitdump/BLHeli #452: Feature request: BLheli_32 output uint64_t value for total Erpm count

FelixvdDonk, 2020-04-15:
Current implementation for measuring Erpm counts the number of commutations of the rotor over the poles and devides it by the time past to get Erpm. Firmwares on the flight controller can then let the user set the amount of poles the motor has and you can then divide the Erpm by this amount to get actual RPM. 

**I think it would be really useful to have a counter that increments for each commutation** and send this over BLheli_32 serial telemetry. A uint64_t counter takes 48 million years to overflow if a motor spins at 40k rpm. So that shouldn't be an issue. 

**Now the reason why;** If you know the start orientation of the motor on bootup and you know the amount of poles of the motor, you can deduct the propellor orientation from this number by a simple modulo operation. This is very useful for fixed wing when you want to prevent prop strike on the ground before landing. 

Imagine having an indicator in the osd that shows you your prop angle in real time. Then if you come in for landing (and you have motor brake on stop enabled) you can see the orientation in the osd and judge if it is safe to land this way. Flying wings for example could really benefit from this feature.

I know it's a niche thing but given the simplicity I hope its worth considering

sskaug, 2020-04-16:
I understand the reasoning and desire here. However the sensorless commutation may easily slip some cycles during the first commutations of startup. And of course also when the motor is stopped, the ESC will not reliably be able to count commutations, should the prop move.

FelixvdDonk, 2020-04-16:
Ah I was worried that might be the case. Thank you for that clarification

closed 2020-04-16 by FelixvdDonk