bitdump/BLHeli #453: 6s quad desync when high throttle

monki04, 2020-04-16:
I’m using Hobbywing Xrotor 60A esc on my 6s quad with T-Motor pro4 1950kv. High throttle will have desync issue and I assume was some setting wrong in blheli32. Motor timing auto or 16 and PWM 48khz also not working.

sskaug, 2020-04-16:
Can you explain more how the desync happens? Does it happen when you are at low rpm and give full throttle? Or does it happen at high (or full?) throttle?
I have many 6S motors here that I have tried to make desync on the bench, in order to analyze what could happen. But I can not make any of them desync here. I have ordered a sample of the T-motor F60 proIV 1950kv too.

monki04, 2020-04-16:
Both situation also will happen desync. Full throttle the motors make a bad noise and than drop on the ground and sharp turn also will suddenly drop on the ground too. The other 6s quad has same issue with 3b hobby motors 2207/1900kv. Both quad using Hobbywing Xrotor 60a ESC. Thank you for your help 🙇🏻🙇🏻🙇🏻🙇🏻🙇🏻

senpaishurikrn, 2020-10-13:
Hello i was wondering dd you find a solution for this issue im encountering the same issue with the same esc and f40pro4 1950kv motors the only way to solve it was to lower my motor output to 90%

monki04, 2020-10-14:
Finally just set up the motor timing to 29. Will not desync anymore

haslinghuis, 2020-10-14:
Commutation timing can be adjusted in three steps. Low is about 0 degrees, mediumlow 8 degrees, medium 15 degrees, mediumhigh 23 degrees and high 30 degrees.

From: https://forum.flitetest.com/index.php?threads/motor-timing-blheli-suit.18846/

senpaishurikrn, 2020-10-14:
Amazing gonna try that now what do you have demag set for and ramup power? And what is the weight of you quad i have the exact esc and motors

monki04, 2020-10-15:
Only set pwm 48khz and motor timing 29. That issue come from the high KV motors. Follow my set up should be solved desync problem