bitdump/BLHeli #454: 6s Fixed wing desync on mid throttle

FelixvdDonk, 2020-04-16:
Motor: sunnysky V2216 650kv
ESC: Iflight Succex 60A metal fet 

Settings:
[image]

desync manifests itself in motor 'chattering' sound when no propeller is mounted. When a propeller is mounted it sounds worse. Like the esc is 'backfiring', basically powering the coils at the wrong timing.

When going to full throttle the chattering goes away significantly but is not gone

sskaug, 2020-04-16:
I do not have your exact ESC or motor, but I tested an Airbot Wraith ESC with a Sunnysky 2212 980kv motor. With your exact settings ;). And my setup runs just fine. Are you sure your ESC and motor are both good?

FelixvdDonk, 2020-04-16:
Hi Sskaug, this seems to only happen with my lower kv motor. Sunnysky 2216 800kv is fine. The ESC is brand new and the motor has been tested with a Blheli_s previously without issue. 

However this slight 'chatter' in the upper ranges of the throttle I have also seen in some other 2204 motors but to a much lesser degree. Could it be that lower kv motors just exarcerbate this issue, and if so what's the mechanism? 

Given how PWM frequency increase seems to help, is this related to the sentence found in the manual at the PWM paragraph; **Programmable frequency also allows for moving of small but potentially disturbing humps in the throttle response. All ESCs have these bumps, with BLHeli_32 they can be moved in the rpm range, to a place where the system has low sensitivity to them.**

sskaug, 2020-04-17:
In that case you are not describing a desync. What happens at some points in the throttle curve, is that harmonics onf the comuutation frequency match the pwm frequency. And that can give rise to small throttle curve nonlinearities.