bitdump/BLHeli #455: Found no valid ESC configuration (new drone) iFlight405 AIO

jitzejaap, 2020-05-01:
Hi, 

I can't get my new drone to work. Or my esc are broken or they need a flash update. 
But I can't read the ESC. (I do have a battery connected when I'm trying to read)

Board: IFlight SuccesX-A F4-AIO

It's a new BNF drone with DJI Air Unit and a TBS crossfire module. The crossfire and DJI Air Unit are working fine. The FC firmware was on IFLIGHT_405_AIO 4.1.2.
After a couple of tries to connect in BLhelisuite32, I updated that to the newest 4.1.6. 

Also no read of the esc. 

In Betaflight: configuration to DSHOT300 and pidloop to 4kHz.

- I tried it on my MacBook Pro (apple) with the blhelisuite32xm. (32.7
- I borrowed a windows laptop from a friend, and tried it also with blhelisuite (32.7 and another download of 32.6. 

Still no read of the ESC. 

in the log file (on apple and windows laptop) is see this error:
armBLB: General error

It does connect, and I see 4 squares of the ESC, but there are no numbers in. 
When I try to read or check I get the error

_'Found no valid ESC configuration'_
and then no reads for the 4 ESC. 

When I push the flash button:
first the same error, then: 
_Serial interface “m4wFSIntf” connected successfully 
Connection to ESC#1 failed_ 

The write button stays unavailable. 

And when I’m in the motor tab. And I put the motor test mode on. 
I do see it’s connected to DSHOT/3D in the left bottom corner. 

 Can someone help me?

[image]

[image]

[image]

[image]

4712, 2020-05-01:
It looks like, if the ESC's are not connected to the FC board, or they are not connected to the battery.
Do you here initialization beeps of the motors when connecting power?
Did you try to spin the motors using the sliders in the "Motors" tab?

jitzejaap, 2020-05-01:
There are no beeps at all and I can't spin the motors (not in Betaflight and not in BLheli)
The Esc should be on the board (it's a AIO) with the Esc included. I will attach a photo. 

[image]

4712, 2020-05-01:
I guess the nuts do not short the motor wires?
If you do not hear any beeps from the motors when connecting the battery, something seems to be wrong with the hardware connection between battery and ESC's/Motors.

jitzejaap, 2020-05-01:
There is rubber down there, so no shorts. 
I bought it BNF, so did not do the soldering myself. 
But I even checked the ESC with a multimeter on continious. The look all fine. no shorts. 

When I connect the battery (6s) I can reach the FC, I do see the light on the FC react to my arming on my TBS tango. Even when I'm throttle up. I see the LED flashing a couple times. 

It looks like there is only no communication with the motors/ESC. 
I thought the 'no beeping' is because of this. 

Therefore I need to update the Firmware on the ESC, but I can't read them. 
[image]

4712, 2020-05-01:
The each ESC normally beeps at least 3 times on power up, even if only a motor and a battery with appropriate voltage is connected. This is independent from the ESC FW revision.

jitzejaap, 2020-05-01:
I got no beeps at all.  never had any beep when first connected it. 

I'm kinda stuck here.. Do I just need to replace the AIO FC board (with esc) or is this fixable.

4712, 2020-05-01:
Sorry, I don't know either.

jitzejaap, 2020-05-01:
If one of the ESC is broken? Could I (most likely) read the other 3 then?
No non of them are readable.

mikeller, 2020-05-02:
@jitzejaap: If you bought this as a BNF and you do not get any beeps from any motors when plugging in a battery then this means that this is DoA. Before you attempt any homemade repairs that will void the warranty, contact your supplier and arrange for a replacement / refund.

4712, 2020-05-02:
@jitzejaap One more question: Did you unplug everything from the board you can, before connecting the battery and listening for beeps?

jitzejaap, 2020-05-02:
@4712 Yes! I tried that. No beeps. Disconnecting the air unit and also the tbs nano rx. 

And also put all the ports sliders off to try without the uards active. But also no result.

Does it matter that tbs is soldered on UARD 6? Instead of 3? Or don’t the uards interfere with the esc.

4712, 2020-05-02:
No idea about the tbs unit soldering. If you didn't solder anything yourself, as @mikeller stated:
your BNF unit seems to be DAO. His advice again: Before you attempt any homemade repairs that will void the warranty, contact your supplier and arrange for a replacement / refund.

LaurentAubouy, 2020-05-10:
@jitzejaap Hi did you found any solution to your probs i have the same situation here.

jitzejaap, 2020-05-10:
@LaurentAubouy no didn’t fix it yet.
Think I tried everything possible to my knowledge,  But if someone knows what it could be all the help is welcome.

Do you have the same flight controller board and same setup with tbs nano? 

It looks like the esc do not get power or something. (Would be great if it would be a setting or firmware thing, because then it most likely fixable)

I don’t know if as one of the ESC is broken, non of them could be readed? Or that it should show the other 3 esc. 

I think only thing left to do is replace the flight controller board AIO. 

Did you had any luck with a solution? 

Ps. I also tried betaflight 4.2 in the mean time. Also no luck to fix it with that.

LaurentAubouy, 2020-05-13:
@jitzejaap 
I did the same as you but unfortunately my esc refused to be connected to BlHeli32 anymore. 
At my first battery connection, everything seems working well but i realized quickly that i have one motor who doesn't spin properly so I've try to fix it on Blheli. 
I've change few things, no much but after that my esc refuse to work.
So i presume there was a malfunction on it and my attempt to fix the motor issue probably didn't help at all.
My esc was a Airbot Furling32 and my Fc a RevoltOsd
Sorry i don't have any clue for your issue, for me i just send a video to the brand and await a response from them.

tielou, 2020-11-12:
Hi @jitzejaap did you ever fix this?
I have exactly the same issue and it was just there from one day to another without having changed anything.

Will-Jachar, 2020-11-17:
Me too. I was flying. Landed (no crash) wanted to arm again a was unable to. Since then every time i connect a battery the ESC beeps three times but the other two beeps (normal startup is three short and two long) are missing. The rest of the quad seems OK. No shorts anywhere. FC, Rx, VTX works normally. I just cannot spin the motors.

mikeller, 2020-11-22:
@Will-Jachar:
> the ESC beeps three times but the other two beeps (normal startup is three short and two long) are missing.

Telltale sign of the signals not going through from the flight controller to the ESC - check your connections, check that motor output is enabled & correctly configured on the flight controller.

cesnietor, 2021-01-30:
Hello, is there any workaround from this? my case is exactly the same using Mac version and eventually BLHeli32 app fails saying `FormError` not found . I did try reinstalling everythin including my FC firmware but this persists.
I'm using `BLHeliSuite32xm 64 1.0.1.0`. For me they used to do the two cycles of beeps and then when trying to configure them I managed to run each of the motors from the app but suddenly the error reported here started to happen, not being able to find the ESCs. I would really appreciate any help for this.

cesnietor, 2021-01-30:
The issue was fixed by Reflashing the FC with Betaflight (in my case Firmware KAKUTEF7- 4.2.6 [use your FC's own version]). Then  Set the configuration of ESC  on Betaflight/Configuration-> ESC/Motor Features  to DSHOT150. After doing this the BLHELISuite was able to recognize the ESCs again. There might be some issue when doing passthrough that was not allowing communication with FC -ESCs screwing the ESCs. Hope this helps.

Standaa, 2021-06-04:
Same problem here. Tested connections with a multimeter. Betaflight connection works fine. BLheli will never find a configuration though on the 4 ESCs.
I get the three beeps, but that's all. Any recos ?

SuperSucculent, 2021-06-06:
Hey I am experiencing the same problem, but slightly different.  To be clear I am VERY new to this, am currently building my first drone.  I downloaded beta flight, connected to my drone, and then went to motors and got them to spin, super ez.  I then went to blheli to alter the directions of the props to spin in the correct manner, and I got this message: 

[image]

Slightly confused, I then looked up a youtube video on flashing Betaflight firmware onto my FC (My FC is this one here: https://www.getfpv.com/t-motor-mini-f7-stack-f45a-3-6s-blheli-32-4-in-1-esc.html)
I followed a Joshua Bardwell video, and got the job done (I determined my FC target, STM32F7X2, and then flashed using beta flight firmware flasher).  That went smoothly, and then I went back to blheli to get greeted with this:

[image]

Confused and concerned, I went back to Betaflight to attempt to spin the motors, and then I got this message:

[image]

REALLY confused I then determined what motor features my FC supports (which is all of them), and then proceeded to choose DShot150.  I then went back to motor control to spin the motors with no success.  I tried various different motor features, but to no avail (at this point im questioning my existence on this earth).  If anyone could let me know what I am doing wrong, or what I need to do to solve this that would be appreciated (please remember that I am a COMPLETE beginner lol).  Any help is appreciated.

4712, 2021-06-06:
@Standaa, @SuperSucculent: First of all, if you cannot hear the 2 final initialization beeps (so 5 beeps in total) and/or cannot spin the motors through "Motors" tab in Betaflight Configurator, there is something wrong with the FC or hardware configuration.
BLHeliSuite32 will not be able to "see" any of the ESC then.

As long as the "there is no motor output protocol selected" warning occurs, when connecting the FC with Betaflight Configurator, the motor outputs still must be activated. Hint: be sure to click "**Save and Reboot**" after selecting a motor output protocol.
BTW: Version 4.2.7. or higher is recommended for Betaflight.

SuperSucculent, 2021-06-06:
@4712 Thanks so much for your response, I reflashed the 4.2.9 beta flight firmware onto the FC (while the battery was plugged in), and then I unplugged and plugged it back in to only hear 3 beeps.  I then got the message that a motor configuration was needed (As before), I then selected DSHOT600, and then clicked Save and Reboot, but still no progress.  If it helps, this is the FC I have: https://store-en.tmotor.com/goods.php?id=1036.

From some research I am doing, some of my issues seem to be lining up with a wrong target flash, but the link above seems to confirm that I am flashing to the correct target:

[image]

I am not sure if there are some other settings that I need to be fiddling with to alter my FC to function, but I am not to clear as to what those are.  From what @4712 said about the beeps, it may be possible there maybe some sort of error in the hardware configuration (which I think you are speaking in regards to the firmware and such, if I am wrong please correct me), or a problem with the FC itself.  To be clear again I am only getting 3 beeps on startup, and like a dummy, I did not backup my FC's default firmware and configuration because I was getting 5 beeps with the original firmware.  I have done a little online searching to try to find a download for my FC's original firmware, but I have had no luck so far.  Please advise as to next steps. Thanks

4712, 2021-06-06:
@SuperSucculent Regarding https://store-en.tmotor.com/images/20200901/1598939247941276.jpg this should be your selection for BF flashing...
[image]

Standaa, 2021-06-06:
My bad @4712 . Your reply made me double check everything again and I eventually realised my ESC signal was wired to the telemetry port... Just tested and everything works as expected. Thank you very much.

SuperSucculent, 2021-06-06:
@4712 That has completely fixed my problem, everything is working beautifully (in betaflight and in blheli32).  I guess my big mistake was fully trusting the Joshua Bardwell video that told me to put the "version" command in the CLI for the FC, and it returned me the STM32F7X2 value, which was incorrect.  Thanks so much for your help and input, it is greatly appreciated.  God bless

4712, 2021-06-06:
@Standaa, @SuperSucculent great that it works now for you!

FalseRhyme, 2022-02-07:
Same problem as above my GOKU Versatile F4 5-IN-1 1S AIO FC when plugged into bheli nothing shows up and I can only get one motor to spin the rest of the esc's says they don't have any firmware on them. Is there a way to fix this without pulling out an Arduino board and soldering. Is there a flywoo config software or is there special esc firmware?