bitdump/BLHeli #457: Linux problem

pchristyUK, 2020-05-11:
This is not a hardware issue, as the hardware all works fine on a borrowed windows computer! I do not have windows.

System: Slackware64-current. Programmer: FTDI plus mod board (all works fine under windows).

Problem: Linux detects the modified FTDI as ttyUSB0. As a user, I have group RW access to ttyUSB0. When I try and select the USB/COM option under the interface selection drop down, the program instantly crashes with the following error: "Exception EAccessViolation in module BLHeliSuite32xl at 000000000085B7A8.
Access violation at address 000000000085B7A8, accessing address 0000000000000000."

The program complains on opening: "./BLHeliSuite32xl: /usr/lib64/libcurl.so.4: no version information available (required by ./BLHeliSuite32xl)"

However:
"locate libcurl
/usr/lib64/libcurl.so
/usr/lib64/libcurl.so.4
/usr/lib64/libcurl.so.4.6.0"

and: "ls -l /usr/lib64/libcurl.so.4
lrwxrwxrwx 1 root root 16 Apr 30 07:56 /usr/lib64/libcurl.so.4 -> libcurl.so.4.6.0*"

I'm normally oretty good at sorting out permissions issues, but this has me stumped! Any ideas?

-- 
Pete

4712, 2020-05-12:
@pchristyUK
**libcurl.so.4 issue:** please try to use the legacy version (BLHeliSuite32xLinux64_1001_legacy.zip).
But this will not help the Exception/crash, when using the menu bar. I'm not sure what is still missing to get the app running. I hope to find a fix soon, but for the moment it is not running under slackware64...

pchristyUK, 2020-05-12:
Thanks for your quick reply! I don't think the libcurl.so.4 issue is critical. I made a temporary install of Mageia 7 (64-bit) on an old machine. This also gave the libcurl warning, but otherwise ran perfectly. Mageia 7 contains some 32-bit compatibility libraries. Slackware 64 is pure 64-bit. Could this be relevant? Running "ldd BLHeliSuite32xl" does not indicate that anything is missing.

-- 
Pete

4712, 2020-05-12:
@pchristyUK You're right, the libcurl.so.4 issue is no show stopper. The crash is produced by the used menu component. I hope to find a fix soon. Meanwhile please use your Mageia system or do not use the menu and select interface suing the ini file \settings\BLHeliSuite32xl.ini
>[Interface]
>InterfaceType=13

[image]

pchristyUK, 2020-05-12:
Thanks for the update! I'll give that a try in the morning!

pchristyUK, 2020-05-13:
No, that didn't work! Altering the .ini file produced the same crash, whether 12 or 13 was selected. Only 14 works.

-- 
Pete

4712, 2020-05-13:
- Open and close the app.
- Edit the ini file and save it.
- Open the app again and have a look at the title bar.

The crash only takes place when clicking the menu bar, please do not use it.
Have a look at the title bar, which interface is shown there.
[image]

pchristyUK, 2020-05-13:
Open and close the app. Check!
Edit the ini file and save it. Check!
Open the app again and have a look at the title bar. App doesn't open! Crashes instantly:
BLHeliSuite32xl/BLHeliSuite32xl: /usr/lib64/libcurl.so.4: no version information available (required by BLHeliSuite32xl/BLHeliSuite32xl)
Exception EAccessViolation in module BLHeliSuite32xl at 000000000085B7A8.
Access violation at address 000000000085B7A8, accessing address 0000000000000000."

4712, 2020-05-13:
could you try

> [Interface]
>InterfaceType=13
>AutoPortScan=0
>TestPortOnEnum=0

pchristyUK, 2020-05-13:
OK, that's an improvement! The app now opens without crashing, and the correct port is showing in the drop down menu at the top.
However, no port is showing in the port window at the bottom of the page. Clicking connect doesn't work (no port selected), and trying to select a port by clicking on the "port" box throws up the same old error message - though this time, the app itself doesn't crash and stays open.

4712, 2020-05-13:
Thanks, that's a bit tricky. Seems to be crashing if enumerating the serial ports. I will come back with a test tool soon...

4712, 2020-05-13:
Here is a tool to test.
Please post the log.txt file, if it is created.
Thanks
[LinuxTestSerialPortEnum.zip](https://github.com/bitdump/BLHeli/files/4621533/LinuxTestSerialPortEnum.zip)

pchristyUK, 2020-05-13:
[log.txt](https://github.com/bitdump/BLHeli/files/4621645/log.txt)

The correct port is ttyUSB0.

4712, 2020-05-13:
Thanks, beside of the time stamp, your log is binary exactly the same as mine here.
So more likely again an issue of the ComboBox....
Here it crashes when i click with the right mouse button into the white colored text area; but it works when using the gray button on the right side.

[image]

pchristyUK, 2020-05-13:
When I left-click  the gray button to the right, it crashes. Right clicking the gray button does nothing. Left clicking in the text area does nothing, right clicking brings up the "cut, copy, paste" dialog box.

4712, 2020-05-13:
Do you use KDE or anything different?

pchristyUK, 2020-05-13:
KDE5. I can try it under XfCE - give me 10 mins or so...

pchristyUK, 2020-05-13:
Exactly the same under XfCE!

4712, 2020-05-13:
Here I have KDE 4.14.21. I will try to install 5. Which one to choose?

pchristyUK, 2020-05-13:
I'm running the latest Plasma 5. This is not standard on Slackware64-current, but is provided by Eric Hameleers (AlienBob), one of the Slackware developers, until Plasma5 is merged into the mainstream. I should add that Mageia7 also uses Plasma5, so its probably not the issue, but I could be wrong! ;-)
What distro are you running?

4712, 2020-05-13:
slackware64-14.2-install-dvd.iso on V-Box

pchristyUK, 2020-05-13:
Eric does a "live" distro using plasma5 that can run from a pen-drive. You can find the Plasma5 version here: http://slackware.uk/people/alien-slacklive/latest/
(make sure you get the plasma5 version)
He also does a Plasm5 for 14.2, but its about to go off-line because -current has diverged so far.

pchristyUK, 2020-05-13:
P.S. You can also "install" the live version to a hard drive, so should be able to run it in a virtual environment.

4712, 2020-05-13:
I don't understand what's going on here.
I started the live iso in vbox and all is fine (Plasma and XfCE) 

[image]
...

pchristyUK, 2020-05-13:
Very odd! I've tried it on two machines here, my workshop machine (AMD based) and my main desktop (Intel based) with the same results. I'm just downloading the live edition now, and will give it a try shortly. In the meantime, I'm going to try an old laptop I'm repairing, with just a basic install on it.

pchristyUK, 2020-05-13:
It looks as if it might be a fault at my end! I installed the unmodified suite on my repair laptop, which only has a basic install and it didn't crash, but neither did it let me select the correct USB port. On the other hand, when I used the modified .ini file, it did finally let me select ttyUSB0. I haven't tried it with the Neuron ESC yet, and want to try a couple of other things as well.

As I originally mentioned, I'm using a FTDI programmer with a bit of added circuitry (as described here: https://www.modelflying.co.uk/forums/postings.asp?th=142379). I also use the FTDI to program a multi-module, and I'm wondering if there is a clash of settings between the Arduino program and your suite. The repair laptop doesn't have Arduino installed!

Either way, something odd is going on!

Let me play around for a while, and I'll get back to you (may take a few hours!). In the meantime, many thanks for your efforts!

-- 
Pete

4712, 2020-05-13:
Thank you for all the effort so far and let me know, what you find out.

pchristyUK, 2020-05-13:
I'm begining to think this is hardware related, somehow! My workshop machine (where I build my models and set up my radio gear) is about 8 years old and based on an AMD Bulldozer processor. My main desktop is around 4 or 5 years old and Intel I7 based. Both these machines exhibit the errors I've described.

The repair laptop is a broken one I was given - a Dell Latitude 3550, I5 based and probably 6 or 7 years old. This works with the modified .ini file, but not with the original.

I also have a new laptop (PCSpecialists), Intel I5 based and only about 12 months old. This works perfectly with the original .ini file!

All the machines are running up-to-date copies of Slackware64-current with AlienBob's Plasma5 upgrade. All except the main desktop are running the same kernel (5.4.30). The main desktop is running 5.6.7 due to compatibility issues with the TV-card drivers.

Other than hardware differences, I am at a loss to explain why some work and others don't. The old repair laptop does not have Arduino installed, but all the others do. Obviously there are some differences between them in terms of the software packages installed, but all are running the same basic system. I cannot see any permissions issues with the ttyUSB0 port, and in any case I don't see why this would cause a crash.

I'm afraid I'm running out of ideas at this point, unless you can think of something?

-- 
Pete

4712, 2020-05-13:
Arduino software conflict is not likely.
Permissions issues with the ttyUSB0 port also not likely.
TestPortOnEnum=1 in the ini file is most likely critical as it opens all ports.
I wish, I had a machine which crashes on scanning the ports...
If there is time, I will try to get more different machines on hand to try out.
I hope some day there will be a light on this weird issues.

4712, 2020-05-16:
@pchristyUK Could you please try the attached test app with all options?
[image]

Thank you!
[LinuxTestSerialPortEnumV2.zip](https://github.com/bitdump/BLHeli/files/4638584/LinuxTestSerialPortEnumV2.zip)

pchristyUK, 2020-05-16:
OK, tried running it, the window opens, no ports are shown, clicking on the "3 dots" box does nothing, and I can't enter any text into the window.

I'm expecting our daughter soon to celebrate my wife's birthday, so may not be able to help much today. I'll keep going as long as I can, but tomorrow should be better!

4712, 2020-05-16:
OK thanks, if the (...) button click does not show anything, then no serial device was found.
I had the "hope" that it will crash, but it doesn't....
Thank you for testing anyway, have a nice celebration!

pchristyUK, 2020-05-16:
The port was definitely there. Dmesg reported ttyUSB0 immediately after plugging in the FDTI adapter.

Cheers, and thanks for your efforts!

4712, 2020-05-16:
Perhaps you need to move the window before it gets enabled...

pchristyUK, 2020-05-16:
OK, that one does not create a logfile, but presents a crash window: "Access violation at address 0000000000839108, accessing address 0000000000000000."
Hope I got the right number of zeros, couldn't copy and paste it!

4712, 2020-05-16:
Great! 
And now please **first** un-check "Enable check HID device" and then lcik the button.

[image]

pchristyUK, 2020-05-17:
OK, that worked without crashing. It brought up two ports, /dev/ttyUSB0 (correct) and /dev/ttyS0. I selected /dev/ttyUSB0 and it still didn't crash.

4712, 2020-05-17:
Ok, thanks a lot for you efforts! 
Seems to be a HID related issue.
Please check out test app, click buttons 1: to 4: and check for any Access violation message.

[LinuxTestHidEnumV1.zip](https://github.com/bitdump/BLHeli/files/4640052/LinuxTestHidEnumV1.zip)

pchristyUK, 2020-05-17:
1: OK 2: Access violation at 8390F8 (lots of zeros in front as before, and accessing address all 0s as before). 3: Same access violation as 2. 4: "No devices found".
Sorry for the delay! I was washing the car!

4712, 2020-05-17:
Thank you so much!
This hardens my suspicion, issue is due to the used HID component.
Finding a workaround should do the job.
Do you have a FVT Usb linker by chance?

pchristyUK, 2020-05-17:
Er, not that I know of! I have the FTDI one that I use for this application, I have some USBASP programmers for Arduinos, and I have some fdd-lite rev2 adapters for FrSky stuff. I suspect that none of these are what you are referring to!

4712, 2020-05-17:
Thanks anyway!

4712, 2020-05-17:
Please test - hopefully the last time - a fixed version.
If all runs as it should, button 3: will show all HID devices.

[LinuxTestHidEnumFixV1.zip](https://github.com/bitdump/BLHeli/files/4640583/LinuxTestHidEnumFixV1.zip)

pchristyUK, 2020-05-17:
No crashes, and button 3 produces: "VID 0x1241, PID 0x1122, path=/dev/hidraw0
serial="", manuf="NIL!", prod="NIL!""

So looking good! :-)

4712, 2020-05-17:
Great!!! 
Is it a mouse?

Now it is easy to fix the BLHeliSuite32 soon.
Thank you so much!!!

pchristyUK, 2020-05-17:
No problem! I'm benefiting too, remember! ;-)

Looking forward to the next release!

-- 
Pete

pchristyUK, 2020-05-17:
Just so you know, I've tried this last program on the other three machines running Slackware64 and also the one running Mageia 7. None of them crashed. Some showed no HID devices, one showed 3 (!), but there were no error messages or other odd effects.

4712, 2020-05-17:
Now here BLHeliSuite32xLinux64_1001 as fixed beta for testing.
 
[BLHeliSuite32xLinux64_Beta1001.zip](https://github.com/bitdump/BLHeli/files/4641125/BLHeliSuite32xLinux64_Beta1001.zip)

pchristyUK, 2020-05-18:
Success! No crashes, correct port selected OK, and it read the speed controller OK. Haven't tried writing back yet, but I have another (smaller) Neuron arriving - hopefully today - that I will need to set up! I am confident it will work OK, but will let you know when I get the new ESC. I will also check it on the other machines, but now I need to mow the garden (will take a few hours!).

Many thanks for you efforts!

-- 
Pete

pchristyUK, 2020-05-18:
It also works on all my Slackware computers and the Mageia one too!
Excellent!
-- 
Pete

4712, 2020-05-18:
Great that it works now on all your machines!
The new BLHeliSuite32xLinux64_1002 will be out soon, of course with the fix included.
### **Thank you so much for your friendly, patient and effective cooperation!!!**

pchristyUK, 2020-05-18:
On the contrary, thank you for all your efforts with this program! It makes life much easier for folks like me! ;-)

(Sorry about shouting in my last post! Don't know what happened there!)

-- 
Pete

4712, 2020-05-18:
;-) 
BLHeliSuite32xLinux64_1002 is published now.

pchristyUK, 2020-05-18:
Excellent! Downloaded, and will be installed tomorrow!

Thanks again!

-- 
Pete

jtamplin, 2020-05-19:
FWIW, I ran into this same problem on Fedora 31 x86_64 with BLHeliSuite32xLinux64_1002.zip, but BLHeliSuite32xLinux64_Beta1001.zip seems to work ok.

1002 fails at startup without ever displaying anything:

`./BLHeliSuite32xl: /lib64/libcurl.so.4: no version information available (required by ./BLHeliSuite32xl)
Exception EAccessViolation in module BLHeliSuite32xl at 00000000004421DD.
Access violation at address 00000000004421DD, accessing address 0000000000000000.
`

4712, 2020-05-19:
@jtamplin Could you check [LinuxTestHidEnumFixV1.zip](https://github.com/bitdump/BLHeli/files/4640583/LinuxTestHidEnumFixV1.zip) Button 3: ?

4712, 2020-05-19:
@jtamplin ...and did BLHeliSuite32xLinux64_1001.zip work or crash as well?

pchristyUK, 2020-05-19:
1002 working fine here, but will check on other machines later this morning.

-- 
Pete

pchristyUK, 2020-05-19:
Working on Mageia 7 OK as well.

-- 
Pete

4712, 2020-05-21:
@pchristyUK Thanks a lot for the feedback!

nschende, 2021-06-23:
Seems the old version has been removed from the download site now? I cant get version 1025 to run on Fedora 34. lubcurl-gnutls warnings and it refuses to launch.   thoughts? seems this issue hasn't been touched in over a year.

[4 more replies]