bitdump/BLHeli #458: Configurator Not Reading

DCinHSV, 2020-05-14:
Noob here, I did use the search beforehand and don't see the answer. I'm trying to run BLHELI to adjust some rotors, but when I connect the bird and the laptop it will not "READ SETUP". Showing logs everything appears to be connecting as it reads the ESC/Board data. The error log shows the info below. Any tips on what to try?TIA
dc************************************************** from LOG
Running - OS: <strong>Windows</strong>, Chrome: <strong>81.0.4044.138</strong>, Configurator: <strong>1.1.0</strong>Uncaught Error: Not found chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/libraries/q.js 134 29 Error: Not foundat chrome-extension://mejfjggmbnocnfibbibmoogocnjbcjnk/js/flashing_helpers.js:170:29Last used port wasn't saved "yet", auto-select disabled.Connecting to: COM9SERIAL: Connection opened with ID: 1, Baud: 115200Serial port <span style="color: green">successfully</span> opened with ID: 1MultiWii API version <span style="color: green">received</span> - <strong>1.41.0</strong>Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>4.0.6</strong>Running firmware released on: <strong>Sep 1 2019 00:44:18</strong>Board: <strong>KTF7</strong>, version: <strong>0</strong>Unique device ID <span style="color: green">received</span> - <strong>0x4500303436510c34333434</strong>Reading setupsending cmd_DeviceInitFlash 0 0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORsending cmd_DeviceInitFlash 0 1received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORsending cmd_DeviceInitFlash 0 2received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORsending cmd_DeviceInitFlash 0 3received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORReading setup <span style="color: green">finished</span>Reading setupsending cmd_DeviceInitFlash 0 0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORsending cmd_DeviceInitFlash 0 1received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORsending cmd_DeviceInitFlash 0 2received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORsending cmd_DeviceInitFlash 0 3received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORReading setup <span style="color: green">finished</span>Reading setupsending cmd_DeviceInitFlash 0 0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 1 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORsending cmd_DeviceInitFlash 0 1received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 2 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORsending cmd_DeviceInitFlash 0 2received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 3 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORsending cmd_DeviceInitFlash 0 3received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0received cmd_DeviceInitFlash ACK_D_GENERAL_ERROR 0 0,0,0,0ESC 4 read settings failed cmd_DeviceInitFlash ACK_D_GENERAL_ERRORReading setup <span style="color: green">finished</span>sending cmd_InterfaceExit 0 0received cmd_InterfaceExit ACK_OK 0 0SERIAL: Connection with ID: 1 closed, Sent: 482 bytes, Received: 718 bytesConnecting to: COM9SERIAL: Connection opened with ID: 2, Baud: 115200Serial port <span style="color: green">successfully</span> opened with ID: 2MultiWii API version <span style="color: green">received</span> - <strong>1.41.0</strong>Flight controller info, identifier: <strong>BTFL</strong>, version: <strong>4.0.6</strong>Running firmware released on: <strong>Sep 1 2019 00:44:18</strong>Board: <strong>KTF7</strong>, version: <strong>0</strong>Unique device ID <span style="color: green">received</span> - <strong>0x4500303436510c34333434</strong>Reading setupsending cmd_DeviceInitFlash 0 0received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4sending cmd_DeviceReadEEprom 0 112received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0ESC 1 read settings failed cmd_DeviceReadEEprom ACK_I_INVALID_CMDsending cmd_DeviceInitFlash 0 1received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4sending cmd_DeviceReadEEprom 0 112received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0ESC 2 read settings failed cmd_DeviceReadEEprom ACK_I_INVALID_CMDsending cmd_DeviceInitFlash 0 2received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4sending cmd_DeviceReadEEprom 0 112received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0ESC 3 read settings failed cmd_DeviceReadEEprom ACK_I_INVALID_CMDsending cmd_DeviceInitFlash 0 3received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4sending cmd_DeviceReadEEprom 0 112received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0ESC 4 read settings failed cmd_DeviceReadEEprom ACK_I_INVALID_CMDReading setup <span style="color: green">finished</span>Reading setupsending cmd_DeviceInitFlash 0 0received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4sending cmd_DeviceReadEEprom 0 112received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0ESC 1 read settings failed cmd_DeviceReadEEprom ACK_I_INVALID_CMDsending cmd_DeviceInitFlash 0 1received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4sending cmd_DeviceReadEEprom 0 112received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0ESC 2 read settings failed cmd_DeviceReadEEprom ACK_I_INVALID_CMDsending cmd_DeviceInitFlash 0 2received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4sending cmd_DeviceReadEEprom 0 112received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0ESC 3 read settings failed cmd_DeviceReadEEprom ACK_I_INVALID_CMDsending cmd_DeviceInitFlash 0 3received cmd_DeviceInitFlash ACK_OK 0 6,51,106,4sending cmd_DeviceReadEEprom 0 112received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0received cmd_DeviceReadEEprom ACK_I_INVALID_CMD 0 0ESC 4 read settings failed cmd_DeviceReadEEprom ACK_I_INVALID_CMDReading setup <span style="color: green">finished</span>
--
  |

4712, 2020-05-15:
@DCinHSV For BLHeli_32 ESCs you need to use BLHeliSuite32, can be found [here](https://github.com/bitdump/BLHeli/blob/master/BLHeli_32%20ARM/README.md#blheli_32-downloads).

DCinHSV, 2020-05-15:
Brilliant . . . and thank you !  Yes wrong version appears to be the
culprit . . . now if I can figure out how to move from one motor to the
next, I should be able to 'reverse 2 motors', save and proceed with the
project.

Thx again for the assist~
dc