bitdump/BLHeli #463: BLHeliSuite32 missing for Windows on Google Drive

grahamcropley, 2020-06-12:
[image]

Windows version seems to be missing?

It's on Mediafire tho'.
[image]

4712, 2020-06-12:
@grahamcropley thanks for letting us know. The file is still there but flagged for abuse.
It seems Google drive virus scanner reports it falsely as virus infected.

4712, 2020-06-12:
Here is the file... 
[BLHeliSuite32_32713.zip](https://github.com/bitdump/BLHeli/files/4772139/BLHeliSuite32_32713.zip)

4712, 2020-06-13:
After long trial and error, i found that 1 out of 73 av engines, "Cyren", detects "W32/Barys.BO.gen!Eldorado". I did not find further info about it. 
Recompiling with German language resource libraries seemed to cure the false detection. So there now is "BLHeliSuite32_32713a.zip" at google drive with that recompiled exe file.

mikeller, 2020-06-14:
@4712: We've been having similar issues with past releases of Betaflight Configurator, which were flagged as 'malware' by various virus scanners. I think a big part in this is that the absence of a signature with a commercial (i.e. non-free) code signing certificate is resulting in a negative bias being added by a lot of these scanners. In most other settings this would be considered a protection racket...

4712, 2020-06-14:
@grahamcropley @mikeller Thanks, issues is resolved now.

closed 2020-06-14 by grahamcropley