bitdump/BLHeli #464: Tekko32 F3 4in1 (version B) Desync Bidirectional Dshot

MimNanopirate, 2020-06-16:
Hi,

I have an issue with these ESC, when i am at low rpm and give full throttle the quad stop and drop
it's  falling out of the sky like it shuts off.

https://youtu.be/s030Vkzg6Bg

To Reproduce

Turn ON RPM filter (bidirectional Dshot) 
Whit no RPM filter is ok

Setup / Versions

- Holybro F7 HDV and Tekko32 F3 4in1 B;
- DJI Vista
- Motors 2306 1750kV

issue with betaflight 4.1 et 4.2 BLheli 32.7 

Diff all

sskaug, 2020-06-20:
There have been reports of some units of the Tekko32 F3 4in1 B ESC that resets when running 6S. You should contact your vendor about this.

etracer65, 2020-06-20:
@sskaug The Betaflight team is receiving lots of reports of bidirectional DSHOT telemetry errors when running DSHOT600 on these and other F3-based ESCs. The errors increase with RPM. Problem seems to be reduced when running DSHOT300.

Additionally there are many reports of intermittent protocol detection problems with the ESCs failing to initialize, or the initialization is delayed and out-of-sync between the motors. Seems to be related to F3-based ESCs (not one particular model/brand). Problem only seems to be related to inverted DSHOT detection.

Taken across all the reports it seems to be some generalized problem with inverted DSHOT handling on the F3-based ESCs. Disabling bidirectional DSHOT and using the normal non-inverted DSHOT seems to resolve the problems.

sskaug, 2020-06-21:
@etracer65 Thanks for the input. We will try to understand what is is going on here.

hydra, 2020-07-07:
This is probably related:

https://github.com/betaflight/betaflight/issues/9886#issuecomment-655085419

The ESC is sending telemetry responses too soon before the FC has had a chance to transition the GPIO pin to INPUT mode.

ChrisRosser, 2020-07-12:
I have the exact same issue. I found that behaviour significantly improved with timing set to auto, PWM to 48kHz (lowest possible for this ESC) and demag to high. I am happy to help test fixes for this issue.

etracer65, 2020-07-12:
@ChrisRosser Disable bidirectional DSHOT as a workaround for now.

hydra, 2020-07-15:
@etracer65 There appears to be an issue with the FC not sending any DSHOT commands at all. See @jhartley 's comments here https://github.com/betaflight/betaflight/issues/9886#issuecomment-657829904

ChrisRosser, 2020-07-15:
Issue still persists with 32.71 beta firmware for Tekko32 F3 4in1 B. It looks to be an issue with 96kHz PWM. No deathroll on 48kHz PWM frequency.

ChrisRosser, 2020-07-15:
@MimNanopirate I'm finding that I don't have a problem if I set the PWM to 48kHz and Demag to high. I hope this helps you too as we're running the same stack.

bluehallu, 2020-07-16:
I have the exact same issue @ChrisRosser describes with the same ESC. This behaviour of some motors not spinning at all after a rearm but fine after reboot is not something I've ever encountered or heard of before. Motor timing doesn't seem to affect it (Tried auto and 23, eventually happened on both) but 96khz definitely brings up this amongst other problems. On 96khz this is almost guaranteed to happen on the first punchout if not before, but aside from this the quad just flies terrible, lots of oscillations and loss of authority, even the motors make a strange noise.

albertmonzo, 2020-07-17:
Had a few more flights. I noticed the issue is much more likely to occur when the battery is full. Trying to do a "punch to the moon" with a fully charged 6S is a reliable way to reproduce. Once the battery is down to 80% or lower it takes a lot of tries to bring out the issue.

bluehallu, 2020-07-17:
Had a few more flights. I noticed the issue is much more likely to occur when the battery is full. Trying to do a "punch to the moon" with a fully charged 6S is a reliable way to reproduce. Once the battery is down to 80% or lower it takes a lot of tries to bring out the issue. Can provide blackbox logs if it helps.

ChrisRosser, 2020-07-17:
@bluehallu Thanks for reporting that you are having the issue too. I happen to have a just received a new tekko32 with separated control. I'll build it into a quad in the next fortnight and we can see if it does the same thing! I haven't had any issues on 48kHz PWM. Do you have any issues on 48kHz?

bluehallu, 2020-07-17:
Yeah, as I said above I get problems even at 48khz. Holybro and they finally agreed to replace both ESCs (I hope with the new version) so I was quite happy to just be careful in the meantime but I just came back from the field and it was... bad. The ESC set itself in flames mid flight charring everything around it and puffing my battery... My advice to anyone with this ESC is to never run on it 6S or get it replaced.

ChrisRosser, 2020-07-17:
Whoa! That escalated quickly! Were you running the 1000uF cap?

bluehallu, 2020-07-18:
Yes I was

bluehallu, 2020-07-22:
So, as I had two of these ESCs (I built two identical quads), I'm on holiday and while I wait for HolyBro to send me the new ones I thought I'd keep trying things with the other one so I spent the whole day at the field today. I tried demag on high, turning off bidirectional Dshot, dshot 300, and even swapped wires on reversed motors so that no motor had to be reversed in blheli. 

None of this helped and it would still occasionally fall off the sky with the usual "motor doesn't turn back on until power cycle". When in this state and before power cycling, rebooting the FC doesn't help, trying to spin the motors from motor tab doesn't do anything and indeed Blheli32 app doesn't even detect the ESCs. I also noticed that sometimes more than one ESC would get in this state, and the ESCs this happened on seemed totally random. I saw it happen at least once in all 4. 

Now the best part is that in the end after a full deflection yaw it fell from the sky again but this time it never recovered. I found it in the grass making a strange noise (video https://youtu.be/JRJi9tedTV4), and now after powering it either makes that noise after the initial startup tones or only some motors spin and others either don't move or the propeller just shakes but doesn't rotate.

ChrisRosser, 2020-07-23:
@bluehallu So I have done some testing with the new version of the Tekko32 F3 and can confirm for me it behaves the same as the old version. Flies poorly on 96kHz with loss of authority at zero throttle. Flies better on 48kHz with some minor loss of authority after full throttle punch and chop, particularly with a full pack. I need to test with bidirectional dshot turned off.

ChrisRosser, 2020-07-23:
@bluehallu OK so I looked at your video and we are running almost IDENTICAL rigs. Same frame, motors, stack, air unit. You seem to be having worse problems than me though. So, how are you mounting your capacitor? I have it mounted just behind the camera with two pieces of motor wire to connect to the battery pads, are you doing something similar?

I'm running BF 4.2.0 filter sliders at 2.0, RPM filter on, dynamic notch set at 0 width percent, Q=250, Low 70Hz, High 350Hz. Everything else at defaults (incl. rates) do you run much higher rates?

bluehallu, 2020-07-23:
@ChrisRosser I've tried default pids and filtering to no avail. Capacitor is mounted directly to the battery pads and held with TPU holder between standoffs. Motors are SmooX 2306/5 1800kv, packs aren't HV. If you're saying the new version has the same behaviour I'm bailing on Holybro. I've just bought a mamba F50 pro and won't ever be buying anything Holybro again.

Bwsbmw, 2020-07-23:
Does anyone else have the tekko32 f3 45amp newest revision with the extra + and - above the tekko32 logo.  running tekko32_f3_C_32.7 firmware. I know that 32.7 is the latest stable release. If I go into flash, I see 32.7(sv) what is the (sv)? I know there has been issues with bidirectional dshot, which I intended to run, but am not sure its been addressed. I would like to run it, as I do on my 2 other Blheli32 quads, but not if I'm going to get desyncs. I will just keep bidirectional dshot off for now unfortunately.

hydra, 2020-07-23:
@ChrisRosser Perhaps you can send your HolyBro ESC to @sskaug for investigation?

4712, 2020-07-23:
@Bwsbmw "32.7(sv)" simply means, that the hex file is not yet packed into the offline firmware resource, but will be downloaded while flashing.

Bwsbmw, 2020-07-23:
Thank you, for that. So is the 32.72 hex have the corrections for the F3
and bidirectional dshot?

ChrisRosser, 2020-07-25:
@hydra Happy to arrange a tekko32 for @sskaug to look at. @sskaug how would it be best to get one to you?

sskaug, 2020-07-25:
If you can email me at sskaug69 at gmail dot com, you can have my address

Bwsbmw, 2020-07-26:
> I have the exact same issue @ChrisRosser describes with the same ESC. This behaviour of some motors not spinning at all after a rearm but fine after reboot is not something I've ever encountered or heard of before. Motor timing doesn't seem to affect it (Tried auto and 23, eventually happened on both) but 96khz definitely brings up this amongst other problems. On 96khz this is almost guaranteed to happen on the first punchout if not before, but aside from this the quad just flies terrible, lots of oscillations and loss of authority, even the motors make a strange noise.
> 
> I have contacted Holybro and received no reply. I find it very suspicious that this ESC has been pulled out of the market and a [new version has been introduced ](http://www.holybro.com/manual/Holybro_Tekko32_F3_4in1_ESC(45A)_spec_v1.2.pdf) with only one difference:
> 
> > The PCB design uses a design in which the drive circuit and the control circuit are separated. This
> > design broadens the overcurrent capability of the drive circuit, and at the same time reduces the
> > electrical interference to the control circuit.
> 
> In my view, electrical interference doesn't manifest itself as a loss of performance but rather as a catastrophic failure like we're seeing. If Holybro is aware that the product had electrical interference problems they should recall it.

I have the brand new revision tekko32_f3_C with the upgraded blah blah blah. I have yet to do anything but hover test in the front yard since I just finished it. Running 32.7.2 bidirectional dshot is off for now. I totally agree, I didn't read about this till after I had ordered it from Holybro. I have 17 more days to contact them if I habe any issues. Timing auto(default), ive set it to 48khz. I dont dare turn bidirectional dshot on yet. I will be doing a full flight tomorrow, and see what it's flying like. I will try bidirectional dshot after I know it's flying ok with it off.

sskaug, 2020-07-27:
We have learned that there is a hardware issue with the B version of the Tekko F3. You can actually see this also from
the second post here. There is a problem for 6S with some motors. So really there seems nothing we can do with FW or settings,
and the proper way of handling it is to return the malfunctioning units to the vendor. Many units have been returned already.

hydra, 2020-07-27:
@sskaug Do you know what the nature of the hardware problem is?  I'm curious.

ChrisRosser, 2020-07-28:
@Bwsbmw If you're running 4S you are unlikely to have any problems. If you are running 6S I would suggest sticking to 48kHz PWM. I found the RevC still had issues at 96kHz on 6S.

sskaug, 2020-07-28:
@hydra. I do not know specifically in this case. But PCB design for ESCs is challenging - during acceleration situations like punchouts or roll stops, motor currents can reach hundreds of amps. Per ESC channel! Such huge current flows in the ground plane can cause disturbances to ESC signals - bemf sensing signals, MCU to driver signals, etc.

ChrisRosser, 2020-07-28:
@Hydra @sskaug My understanding is that higher PWM frequencies create more electromagnetic interference (EMI) (it increases with switching frequency) as do higher voltages (due to higher inrush currents etc.). It seems like a fully charged 6S pack going from 0 to 100% throttle instantly is a perfect storm for EMI. I imagine this overwhelms some part of the motor control circuit and renders it unable to commutate the motor correctly causing the quad to fall out of the sky.
In the new RevC design the control electronics has been moved onto a separate board with what looks like a ground plane in between. This should attenuate electromagnetic interference significantly and protect the control electronics. They've also upgraded the BEC, presumably to guarantee a stable supply voltage to the control circuit even during extremely challenging supply voltage conditions.

Bwsbmw, 2020-07-28:
Chris, no I havent had any issue with Bidirectional Dshot off and 48khz,
but I had set it to 48khz and have not flown it on 96khz yet. I'm going
tomorrow morning, if I try 96khz dshot off on 4s, what should I look out
for? Also I would love to try bidirectional Dshot on, and 48khz since the
esc is already set their, if you don't think their will be any issues on
4s, rcinpower smoox 2306.5-2580kv. Really would like the rpm filtering. I
can try 96khz, bidirectional off, I just really dont want to have a desync.
Its flys solid dshot600, 32.7.2 default, timing auto(default), only the 96
to 48khz I changed.

ChrisRosser, 2020-07-29:
@Bwsbmw I think you should be fine on 4S to try 96kHz. Look for wobbles and bobbles when chopping the throttle. Then punch out from 0 - 100% throttle. Be ready to disarm if the quad desyncs. Do this low to the ground over soft ground (long grass) so that if a desync happens the quad has a soft landing. You should definitely have rpm filter on if you can it really helps reduce motor noise in the gyro.

Bwsbmw, 2020-07-30:
> @Bwsbmw I think you should be fine on 4S to try 96kHz. Look for wobbles and bobbles when chopping the throttle. Then punch out from 0 - 100% throttle. Be ready to disarm if the quad desyncs. Do this [...]

I was able to fly bidirectional Dshot today, 4s on the Tekko32 F3 rev C. Rpm showed 0.00% error on all 4 motors pre flight. It flew fine on 48khz, 32.7.2 doing multiple punchouts, then to quick snap rolls, opposite back to back. I was unable to test 96khz today as I forgot a otg cable. Only thing that was odd was esc temp that is displayed in the OSD was stable say 100F the other day with bidirectional off. Turn on bidirectional dshot, and the numbers jump around fast, numbers between 70, and the real temp I assume 100F. It went up to 2XX for a split second a few times. I dont know of that would activate the 140c protection in blheli32suite. Their was one 0-100 punch out were motor noise on HD footage seamed to raise, flatten for a split second, then raise quickly again to full.  I dont have a log of that event, I was not going straight up. I was swooping down then going back up. I checked my dvr and I was at 100% throttle.  Didnt hear it any on other punch out, or any of the other 4 flights. Thing I really noticed so far was the erroneous esc temp jumping around on OSD with bidirectional dshot on. Motor noise was much better, most on a log was 201hz on pitch Gyro_Scaled, logs looked much better on default pids and filters BF 4.2 kakutef7miniv2, dshot600, 8k8k, 5.5 %idle.

ChrisRosser, 2020-07-30:
@Bwsbmw Glad to hear it all went well!  There is a potentially related issue here: #465 I would suggest disabling ESC telemetry while running bidirectional DSHOT to reduce load on the ESC MCU and therefore reduce the likelihood of late packets. Is there any critical information you need from ESC telemetry?

Bwsbmw, 2020-07-30:
> @Bwsbmw Glad to hear it all went well! There is a potentially related issue here: #465 I would suggest disabling ESC telemetry while running bidirectional DSHOT to reduce load on the ESC MCU and [...]

Hey Chris, To be 100% honest, no, I'll dont use esc telemetry for anything. Well unless it's has to be on for bidirectional dshot to work, but im pretty sure it doesnt. I was just monitor the esc temp on the first flights. i know it runs at about 105F on a hot day now, so no need for it.  The esc sends current data adc anyways, so that's the only info I use from the ESC I use is for Mah consumed. To turn off telemetry, are you reffering to under the ESC/Motor Features, on the configuration page of BF, that says ESC_SENSOR (uses KISS/Blheli32 telemetry over a seperate wire)? Im asumming thats what you may be refering to, because it is on. There's also the Telemetry under other features further down but thats different, not for the esc. Thats for my crossfire I believe. As of right now, the only thing I monitor from the esc is current consumption via adc. So telemetry can be turned off. Thanks for the recommendations.

ChrisRosser, 2020-07-30:
@Bwsbmw  You are correct. ESC_SENSOR (uses KISS/Blheli32 telemetry over a seperate wire) is what you need to turn off. Happy flying!

Bwsbmw, 2020-07-30:
> @Bwsbmw You are correct. ESC_SENSOR (uses KISS/Blheli32 telemetry over a seperate wire) is what you need to turn off. Happy flying!

Thank you, you as well. I just have a kinda quick question thats been in the back of my head.. This specific quad ive been refering to, dshot_burst=off(default)Kakutef7mini 4.2 target. Stays off after bidirectional dshot is enabled. My other 2 quads with rpm filter dshot_burst=on(default)omnibusf4sd target 4.2, and matek405ctr also burst=on(default)4.2 Never changed anything besides turn bidshot on.  What does that even do, and should it be on, off, or doesnt matter? If you don't mind me asking? I dont plan on touching anything, as all 3 are flying with rpm filtering enabled. Just interested, as I dont know.  Thanks again.

MimNanopirate, 2020-08-01:
Holybro recognized the issue with bidirectional Dshot.
They propose me to change the ESC.
Do you know if they really fixed the problem with a new version ?
I wouldn't like to end up with the same ESC 😆

Bwsbmw, 2020-08-02:
I'm running the new version 30x30 on 4s, and am not having any issues. Although I have not tried 96khz, I changed it to 48khz when I upgraded the tekko32 f3 45amp rev C ESC to 32.7.2 and its flying beautiful. You will know its the new version, it will have a extra + and - on the top side if the board under the current sensor. You can use it for an extra cap, or wire a vtx that takes battery voltage which is what I did. The pictures of the new esc are on the Holybro website. 
I'm sorry I cannot speak for 6s, as I dont run 6s. Which is where I think the problem likes to show itself. It's not believed to be esc specific, but F3 related. This version seams to work, ive got over 16packs threw it, and no desyncs. On 4s.

MimNanopirate, 2020-08-02:
Ok thank you, cross fingers for 6s, someone tested 6s ?

ChrisRosser, 2020-08-02:
@MimNanopirate Running 6S on the new version C board @ 48kHz and its fine! Make sure you use the included Cap on the ESC battery pads. Happy flying!

MimNanopirate, 2020-08-02:
Good news, thank you !

Bwsbmw, 2020-08-02:
> Good news, thank you !

I forgot to mention, you may want to check the 4in1 connector. When mine came, and I checked the connector with a uv light. Sure enough pins were also oversprayed with the conformal coating they use. Dont know how much this would really effect it, if any but I figured id clean it off. 99% isopropyl or 91% Its a very light coating, but If you do it, be very very careful of the pins.

rgmorales, 2020-08-11:
Is there are chance this issue is being resolved any time soon? Lots of issues on F3 ESC, I had the same issues with a Tekko32, but it does not work even with 48kHz. I do not want to disable DSHOT... 

Thanks!

SnowLeopardFPV, 2020-08-11:
I would also like to know if we should be sending our Tekko32 4in1 ESCs back for a refund or holding out for some kind of firmware fix. I have one of the latest revision Tekko32F3 Metal 4in1 65A ESCs (the one [HERE](http://www.holybro.com/product/tekko32f3-metal-4in1-esc65a)) which isn't yet soldered into my quad but after reading this, fitting it to my brand new build doesn't exactly fill me with confidence.

It's not clear if my ESC is affected by this issue (it's running the TEKKO32_F3-4in1_B BLHeli32 ESC layout firmware), but I need to decide soon if I should use a different non-F3 based ESC on my build or just wait for a confirmed and tested fix which eliminates all of the issues we are seeing reported here.

Does anyone have any further news?

Bwsbmw, 2020-08-11:
I am running the Latest Tekko32 45amp f3 rev C, 32.7.2, no issues with
bidirectional dshot on 48khz 4s. I am not sure of the 65amp version, I've
only been able to confirm its working on the rev C.

SnowLeopardFPV, 2020-08-11:
Does anyone know how we tell what revision of board we have? I don't see a revision number stated anywhere on my Tekko32F3 Metal 4in1 65A ESC board. Does the "_B" in the firmware layout name on mine indicate the revision (TEKKO32_F3-4in1**_B**)?

Mine will also be running on 6S which worries me even more because from what I'm reading, the issues seem to be more prevalent on 6S compared to 4S.

ChrisRosser, 2020-08-12:
> I would also like to know if we should be sending our Tekko32 4in1 ESCs back for a refund or holding out for some kind of firmware fix. I have one of the latest revision Tekko32F3 Metal 4in1 65A [...]

Snowleopard, looking at the photo you sent can you confirm you have an ESC with a separate board on the back? (a board on board design)

The board on board design of the new ESCs reduces interference and seems to prevent the falling out of the sky behaviour seen with the older single board version of the tekko32.

If you have board on board I would be confident in saying you are ay-OK. If you DO NOT have board on board you should get a replacement new version with board on board before running 6S.

SnowLeopardFPV, 2020-08-12:
@ChrisRosser, thanks for the clarification. Yes I can confirm that I have the very latest version of the Tekko32F3 Metal 4in1 65A ESCs with the separate control daughterboard attached to the main board. Photos of my actual ESC are at the links below:-

https://i.imgur.com/Lk09qLr.jpg
https://i.imgur.com/rWH6610.jpg

Is 32.7.2 now available as a release build or is it still in testing, and does it solve all of the known F3 based ESC issues? The last time I checked I only saw 32.7 in the list of available versions within BLHeliSuite32 for the TEKKO32_F3_4in1_B firmware layout which is the same as the version already installed on the ESCs. If 32.7.2 is still in beta, then rather than manually flashing the test HEX file, I will probably just wait because the ESC on my new build isn't even soldered up yet so there will probably be no benefit in me installing a test version until my quad is actually flyable, by which time 32.7.2 may have made it into a final release.

On a side note, I know it doesn't solve the problem at hand, but do the F3 based ESCs work fine without issues on the 32.6 release? Obviously there is no bi-directional DSHOT capabilities in that older version, but issues with the F3 based ESCs only seem to have recently reared their ugly head, and F3 based ESCs have been around for quite some time.

ChrisRosser, 2020-08-12:
@SnowLeopardFPV  With the latest version you should be fine. Use the provided capacitor on the battery pads and SEND IT!

32.7.2 is still a beta version. I'm running it on 3 quads with no issues and would recommend using it. It fixes a very specific issue with F3 ESCs when they are running bidirectional DSHOT. F3 ESCs are so fast that they were able to send the DSHOT response frame before the flight controller was ready to receive it causing errors. 32.7.2 addresses this problem see #468 for more detail.

32.6 doesn't support bidirectional DSHOT so there isn't a problem as the ESC never tries to send response frames. 

Regarding issues with F3 ESCs. I think there are two separate issues the bug described in #468 which has been fixed in 32.7.2 and hardware related issues with early versions of the Tekko32 F3 ESC. This has understandably led to some concern over F3 ESCs but I think your ESC is unaffected by the hardware issues so no worries there.

My recommendation to you is to manually flash 32.7.2, turn on RPM filtering, and enjoy. It's too good a feature to pass up! You shouldn't have any issues in my experience.

rgmorales, 2020-08-12:
Great to know... for the unlucky ones with the older versions (like me) we just have to throw our ESCs in the garbage...

ChrisRosser, 2020-08-12:
@rgmorales I would contact Holybro for replacements. In my experience they have good customer service contactable at: productservice@holybro.com. Reference this thread as they are aware of it.

I feel for them as I think these ESCs are OEM'd by Airbot so they are making the best of a bad situation.

rgmorales, 2020-08-12:
I spoke to Holybro, they want customers to pay U$25 shipping for them to send replacements... That is half of the cost of a new ESC.  I simple don't believe they did not test this enough to send it to the market. A simple quad build with bidirectional DSHOT enabled would be enough to test if this works, and would have saved us from crashing expensive gear due to their mistake. I crashed U$100 motors, props, DJI antennas and a brand new frame due to this. And I also lost money on the ESC. 
It is inconceivable that the customer has to lose so much money for their mistake and still have to pay more for shipping to fix their mess.
This is how you damage your brand. I'm done with them.

ChrisRosser, 2020-08-12:
@rgmorales

I'm in the UK and got mine sent free from Germany. If you are Europe you might be able to ask for the same deal? I would also make that case to them as it's a fair comment and you shouldn't have to pay anything for a replacement (you may also have some luck raising this with the shop you bought it from?). As for the rest of what you said, it's true and fair to say it. They aren't the first company to release a flawed product and they won't be the last. No company is immune in my experience no matter how expensive a brand they are. I hope they read this and realise that if they handle this well they can get a reputation for good customer service by helping you out!

rgmorales, 2020-08-12:
I'm in the US, and the store I bought it will not be able to replace it for me. I exchanged so many emails with them and still the same. 

I already replaced the ESC by a XRotor, and will not use this again.  

Sad thing is, I was planning to buy a high end AIO with metal ESCs from them, now I cannot trust this brand anymore.

rgmorales, 2020-08-13:
Holybro is fixing my case.

Question on the AIO unit with metal ESCs (Kakute F7 HDV AIO Combo with 65A ESCs https://shop.holybro.com/kakute-f7-hdv-aio-combo_p1193.html) any ideas if those ESCs have the same issue? They are F3 ESCs... 

Thanks!

SnowLeopardFPV, 2020-08-13:
@ChrisRosser, thanks for the confirmation. After what you have said I am now feeling a lot more confident that the 32.7.2 fix has very likely resolved the issues. It will be a while before I can SEND IT (blame that on Banggood's slow dispatch of parts), but I think I will still make the maiden flights over an empty field to be on the safe side until I have full confidence that everything is fine.

It would also be good to hear other people's feedback on the 32.7.2 beta firmware, whether that be good or bad.

ChrisRosser, 2020-08-14:
@rgmorales Very glad to hear this news! I haven't heard of anyone having issues with individual ESCs. In an individual ESC arrangement there is a lot less electrical noise because of the greater separation between the drive circuits for each motor. I would encourage you to put a 35V 330uF cap on the power leads to each ESC which will really help keep the noise under control. I imagine Holybro supply suitable capacitors with the kit so you should be good to go. I would suggest 32.7.2 firmware and 48kHz PWM as a starting point. Ramp up power at 25% rather than 50% for a 5in is what I recommend to reduce current spikes during fast RPM changes.

[31 more replies]