bitdump/BLHeli #465: Bidirectional DSHOT and ESC telemetry conflict?

Diaoul, 2020-06-19:
Error rate is more than 0.00% when both bidirectional dshot and esc telemetry are enabled.

This is related to https://github.com/betaflight/betaflight/issues/9929 ans suggested to be reported here.
Please see the details of the configuration on the linked issue.

**Description**
* With bidirectionnal DSHOT enabled and ESC_SENSOR disabled I have no error as I increase the throttle in the Motors tab: 0.00%
* With bidirectionnal DSHOT enabled and ESC_SENSOR enabled I have no some errors as I increase the throttle in the Motors tab: 0.05% to 0.30%

**Setup**
 - FC Matek F722
 - ESC: AIRBOT Wraight32 V2 35A
 - ESC signal wires twisted with ground and connected with the others on UART4
 - Betaflight 4.2.0
 - Betaflight Configurator 10.7.0
 - BLHeli Rev. 32.7

sskaug, 2020-06-22:
Thanks for the information. This is something we will have to investigate.

sskaug, 2020-07-15:
I have now looked at this, and I believe it is related to the delay between the Dshot input and the rpm info response..
Here is a table of measured error rates vs input to response delay (in microseconds, in addition to the intrinsic 22us delay from F051 processing):
[image]
For Dshot600 with telemetry, increasing the delay from the default 8us to 18us gives a 100% error rate.
While Dshot300 for some reaon is much more forgiving.

Running telemetry taxes the MCU more, and sometimes the rpm data response gets delayed due to other tasks being processed. And as the above data shows, sometimes it is delayed more than BF can tolerate.
Increasing the allowed timing window in BF would be the best fix for this.

closed 2020-07-19 by sskaug

Diaoul, 2020-07-19:
Thank you for looking into it. Is it not possible to reduce the delay from the ESC rather than increase the tolerance on BF side? Maybe there is some optimization that could be done on the telemetry part? Or a bug causing those inconsistent delays?

Diaoul, 2020-07-19:
There is some test firmware on the linked issue. If that's of any use I can give it a shot.

sskaug, 2020-07-19:
Note that the Re32.7.2 addresses and issue of too short delay between dshot input and response. While this issue is about too long delay

etracer65, 2020-07-19:
@sskaug 

> Running telemetry taxes the MCU more, and sometimes the rpm data response gets delayed due to other tasks being processed. And as the above data shows, sometimes it is delayed more than BF can tolerate.
Increasing the allowed timing window in BF would be the best fix for this.

Unfortunately this is not likely to be a viable solution. Betaflight has to account for the time it takes to send the DSHOT packet, change the GPIO pin from an output to an input, receive the actual RPM telemetry response from the ESC, then switch the pin back to an output - all before it's time to send the next DSHOT packet. There's not much time to extend this window when running at 8K PID loop for example.

And if you were testing with DSHOT300 but still an 8K PID loop then you were in reality only sending a DSHOT packet every other loop because again the amount of time between was not sufficient to complete all the above.

sskaug, 2020-07-20:
My testing was done at 2k PID, as is described in my post 4 days ago. I can understand that timing can be tight for 8k PID, but it is not at 4k or 2k.

mikeller, 2020-08-02:
> My testing was done at 2k PID, as is described in my post 4 days ago. I can understand that timing can be tight for 8k PID, but it is not at 4k or 2k.

Unfortunately 2 kHz PID loop speed is pretty much what no user will want to use, as it results in flight behaviour that is worse than what their hardware can support - not sure there is much value in testing / fixing configurations that nobody will want to use.

sskaug, 2020-10-26:
FYI - this has now been addressed and solved in the Rev32.7.3 test code.

mathiasvr, 2021-10-13:
@npdgm https://github.com/bitdump/BLHeli/commit/41967a136ba738198f41e53e5b473d0d38819a74