bitdump/BLHeli #466: BLHeli (orig. SiLabs version) with serial control instead of PWM?

AndrewRabbitt, 2020-06-29:
So, clearly I can talk to a SiLabs-based ESC with BLHeliSuite over a serial link but could I setup duplex comms over a serial link that I could send the PWM value across and get an RPM report back in a response packet?

An ambitious project for a newbie but I've already got a TTL speed signal to work with my own hacked version so I'm stepping it up a notch here.  I've had a brief read/browse through the code but I can't see where the BLHeliSuite comms stuff is coded.  Any hints for slow learners like me, Steffen?

AndrewRabbitt, 2020-06-30:
I just discovered what DSHOT is all about - not that I fully understand how the telemetry bit works...

Am I right that the serial comms is all in the bootloader?

mikeller, 2020-07-06:
@AndrewRabbitt:

> Am I right that the serial comms is all in the bootloader?

Depends on what you define as 'serial comms' - the boot loader implements a serial protocol that is used to update firmware and configuration, and the firmware implements the Dshot protocol used to receive speed information and commands and send back RPM values.

AndrewRabbitt, 2020-07-06:
@mikeller Yes, I'm beginning to understand this a bit more.  I'm looking at the original BLHeli code rather than the BLHeli_S or 32 code which already has DSHOT implemented.  This is due to the ESCs (ZTW Spider 30A) I've embedded in my project's design.  I'm thinking that it may be easier to update to a BLHeli_S-based controller than try to update the code.

mikeller, 2020-07-06:
@AndrewRabbitt: If you want digital control and feedback then definitely.