bitdump/BLHeli #467: Configuration of settings using Betaflight code

haslinghuis, 2020-07-04:
@sskaug 
Not sure this is the right way to ask this question.
We like to configure some BLHELI32 settings from Betaflight.
In this case we like to configure Motor direction as would be possible with BlHeli protocol in Betaflight's [code](https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_4way_avrootloader.c).
Could you provide a pointer for this?

sskaug, 2020-07-04:
Unfortunately due to BLHeli_32 being closed source, the bootloader communication protocol is not open.
Besides, doing it from BF via the bootloader will be quite complex, with first establishing communication, then reading the settings section, erasing it and then reprogramming it. 
I assume you are aware of the dshot command protocol - isn't that an easier way?

haslinghuis, 2020-07-04:
We already tried with dshot, but we don't want to reduce use case. Not everyone is using dshot and new protocols are on the corner. Also more configurators are integrating the BLHeli(_32) configurators.

The plan is to use avr rootloader for this: https://github.com/betaflight/betaflight/blob/master/src/main/io/serial_4way_avrootloader.c

I think if BLHeli_32 publishes eeprom settings and protocols and opens the mcu's then they can still keep providing default firmware from the factory and getting a royalty while allowing hobbyists to reflash if they want and developers to create interfaces.

haslinghuis, 2020-07-05:
@sskaug we are aware about dshot protocol specification, although can't find a more detailed datasheet for it, only what's currently in the betaflight code.

I accept the standpoint about close source, it is what it is. Maybe you are able to provide a small API to do some basic things without exposing the protocol (like a reboot of the bootloader from software). Then we can still can change direction in esc (dshot or other) protocol. @limonspb could that be a workaround for you?

limonspb, 2020-07-06:
@sskaug 
i was wondering the other day how could FL1 do this with blheli32 also?
Did they reverse engineer blheli32 bootloader protocol?

A question about Dshot if we decide to go this way:
I was not able to find a full set of commands. In Betaflight code they say that SETTINGS_REQUEST is not implemented.
Could you please point me on where I can check the full list of dshot commands and how they are working?
Thanks!
[image]

maybenikhil, 2020-07-06:
Settings Request isn't implemented in BLHELI_32 either, and @limonspb given my knowledge (this stands true mostly for 4-in-1s, but if the ESCs aren't programmed yet..) I would say it would be safer to assume that the ESC direction is "normal" and then change it.

limonspb, 2020-07-06:
@maybenikhil but then how FL1/Falco configurator are doing that?...

maybenikhil, 2020-07-06:
@limonspb that would be my guess, given that both of them are closed source, which gives the advantage of money to make a deal. But I couldn't say, and I don't know if there is any transparency about it

mikeller, 2020-07-06:
@limonspb: See the comment on `DSHOT_CMD_ESC_INFO` - the reason that `DSHOT_CMD_SETTINGS_REQUEST` was never implemented is that the ESC info response already returns all relevant information in BLHeli_32.

sskaug, 2020-07-06:
I do not know how FL1 does it, but I would presume with Dshot commands. DIGITAL_CMD_SPIN_DIRECTION_NORMAL sets the direction to what is currently programmed in ESC settings in flash. DIGITAL_CMD_SPIN_DIRECTION_REVERSED sets it to the opposite direction.
A simple description of the BLHeli_32 command set can be found here: https://github.com/bitdump/BLHeli/blob/master/BLHeli_32%20ARM/BLHeli_32%20Firmware%20specs/Digital_Cmd_Spec.txt