bitdump/BLHeli #469: Crash on MacOSBig Sur

EnricoBara, 2020-07-08:
I know Big Sur is just at the second beta, I just want to inform you that the suite crashes after opening in this SO.
Thanks

4712, 2020-07-09:
@EnricoBara Sorry, too early to say anything about crashes and or functionality here.

EnricoBara, 2020-07-09:
I know I know

vanawaker, 2020-07-31:
Still crashing on Big Sur, and all the virtual machines are crash, too. My Mac now even can't change the motor direction lol.

AndreaCCIE, 2020-11-12:
Today Big Sur is released, the issue is still there

Martivip, 2020-11-13:
I have the same issue with official production macOS Big Sur 11.0.1

4712, 2020-11-13:
Sorry, still waiting for development environment code patch for Big Sur.

4712, 2020-11-14:
Please try BLHeliSuite32xm_MacOS64_1009_Beta_BigSur.zip.

Martivip, 2020-11-14:
Thank You works as is should 👍

ildeuz, 2020-11-15:
sorry but where is 
BLHeliSuite32xm_MacOS64_1009_Beta_BigSur.zip
yuo have a link,please?

ildeuz, 2020-11-15:
> Please try BLHeliSuite32xm_MacOS64_1009_Beta_BigSur.zip.

where is this file? I have searched but cannot find it. do you have links?

AndreaCCIE, 2020-11-16:
THanks, I can confirm it does work on mine too.

joonho0724, 2020-11-25:
BLHeliSuite32xm_MacOS64_1009_Beta_BigSur 어디에서 다운 받나요

joonho0724, 2020-11-25:
감사합니다

Pairan, 2020-11-26:
doesnt work for me! It keeps opening and closes immediately!

closed 2022-07-14 by EnricoBara