bitdump/BLHeli #470: Tekko32 F3 4 in 1 Rev B Corrupted settings in BLHELI_32 Android app

ChrisRosser, 2020-07-12:
No problem with BLHELI_32 Suite on Windows but cannot edit ESC settings using the BLHELI_32 Android app.

App connects OK but comes up with corrupted settings warning when I try to read the ESC data. No similar warning when using BLHELI32_Suite on Windows. App tells me to reflash ESCs but this does not fix the issue.

I suspect (perhaps naively) that it may have something to do with the 96kHz PWM frequency of this ESC. Perhaps this is higher than the android apk is expecting and so it gets upset?

sskaug, 2020-07-12:
Thanks for reporting - you are probably right. We'll look into it.

sskaug, 2020-07-19:
An updated app is now published and should be available from Google Play within a few hours

closed 2020-07-19 by sskaug